kiddin9/Kwrt #2: 能不能改成本地版的？

lx521603, 2020-04-09:
老大，你这个牛啊，但能不能改个本地版的，云编译太费时间了，要2小时。

本地几分钟就编译好了。

garypang13, 2020-04-09:
```
git clone https://github.com/openwrt/openwrt
git clone https://github.com/garypang13/Actions-OpenWrt-Nginx
cp -Rf Actions-OpenWrt-Nginx/* openwrt/
cd openwrt
./scripts/feeds update -a && ./scripts/feeds install -a
../diy.sh
cp -Rf diy/* ./
mv X86_64.config .config
make defconfig
```
在本地按照这个步骤执行就好了.

lx521603, 2020-04-09:
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'python-light', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
tmp/.config-package.in:125199:error: recursive dependency detected!
tmp/.config-package.in:125199:	symbol PACKAGE_wrtbwmon depends on BUSYBOX_DEFAULT_IP
package/utils/busybox/Config-defaults.in:2340:	symbol BUSYBOX_DEFAULT_IP depends on PACKAGE_busybox
tmp/.config-package.in:130:	symbol PACKAGE_busybox is selected by PACKAGE_wrtbwmon
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:119: recipe for target 'defconfig' failed
make: *** [defconfig] Error 1

云编译不成功，每次都在这里出错。

我试试本地编译，谢谢指导。

lx521603, 2020-04-09:
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
sed: -e 表达式 #1, 字符 38: 未终止的“s”命令↵
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
./diy.sh: 行 71: curl：未找到命令
basename: 缺少操作数
Try 'basename --help' for more information.
find: 警告: 不赞成使用 -d 选项；请使用 -depth 代替，因为后者更符合 POSIX 特性。
xargs: rename: 没有那个文件或目录

执行diy.sh后的错误。

lx521603, 2020-04-09:
li@aki:~/op$ make defconfig
ERROR: please fix package/feeds/luci-app-dockerman/Makefile - see logs/package/feeds/luci-app-dockerman/dump.txt for details
ERROR: please fix package/feeds/telephony/kamailio-5.x/Makefile - see logs/package/feeds/telephony/kamailio-5.x/dump.txt for details
Collecting package info: done
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'python-light', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
tmp/.config-package.in:125199:error: recursive dependency detected!
tmp/.config-package.in:125199:  symbol PACKAGE_wrtbwmon depends on BUSYBOX_DEFAULT_IP
package/utils/busybox/Config-defaults.in:2340:  symbol BUSYBOX_DEFAULT_IP depends on PACKAGE_busybox
tmp/.config-package.in:130:     symbol PACKAGE_busybox is selected by PACKAGE_wrtbwmon
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

/home/li/op/include/toplevel.mk:119: recipe for target 'defconfig' failed
make: *** [defconfig] Error 1
li@aki:~/op$

这个是执行 make defconfig后的

lx521603, 2020-04-09:
li@aki:~/op$ make menuconfig
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'python-light', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
tmp/.config-package.in:125199:error: recursive dependency detected!
tmp/.config-package.in:125199:  symbol PACKAGE_wrtbwmon depends on BUSYBOX_DEFAULT_IP
package/utils/busybox/Config-defaults.in:2340:  symbol BUSYBOX_DEFAULT_IP depends on PACKAGE_busybox
tmp/.config-package.in:130:     symbol PACKAGE_busybox is selected by PACKAGE_wrtbwmon
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

/home/li/op/include/toplevel.mk:134: recipe for target 'menuconfig' failed
make: *** [menuconfig] Error 1
li@aki:~/op$

尝试make menuconfig 不成功，不能继续。

garypang13, 2020-04-09:
本地编译前需要先安装好环境

命令行输入 sudo apt-get update ，然后输入 sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler ccache xsltproc rename antlr3 gperf

至于云编译失败可能是github的问题,明天再试试,我在本地试了没问题的.

lx521603, 2020-04-09:
但我本地编译其它的没问题啊，我今天还编译过lean的R20.4.8版本。lienol的dev-master分支也在更新。
唯一的区别是我clone到了op目录，因为openwrt和lionel大的名字冲突了。

garypang13, 2020-04-09:
已修复
删除原来的,重新运行本地编译脚本
```
rm -Rf openwrt Actions-OpenWrt-Nginx
git clone https://github.com/openwrt/openwrt
git clone https://github.com/garypang13/Actions-OpenWrt-Nginx
cp -Rf Actions-OpenWrt-Nginx/* openwrt/
cd openwrt
./scripts/feeds update -a && ./scripts/feeds install -a
./diy.sh
cp -Rf diy/* ./
mv X86_64.config .config
make defconfig
```

lx521603, 2020-04-10:
fullconenat.patch   100%[===================>]   2.02K  --.-KB/s  用时 0s

2020-04-10 13:36:11 (24.6 MB/s) - 已保存 “package/network/config/firewall/patche                        s/fullconenat.patch” [2071/2071])

./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
sed：-e 表达式 #1，字符 38：未终止的“s”命令
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
./diy.sh:行72: curl：未找到命令
basename: 缺少操作数
请尝试执行 "basename --help" 来获取更多信息。
find: 警告: 不赞成使用 -d 选项；请使用 -depth 代替，因为后者更符合 POSIX 特性。

这次能执行完diy.sh，但有这些信息，我编译看看能不能完成。

早上云编译了2次，能编译完成，但有点问题。有一大半的选项，点进去是一直在转圈，显示“正在载入视图…”。

garypang13, 2020-04-10:
./diy.sh: 行 71: curl：未找到命令
确认linux中有没安装curl
sudo apt-get install curl

lx521603, 2020-04-10:
touch /home/li/openwrt/staging_dir/target-x86_64_musl/stamp/.v2ray-plugin_installed
make[3]: Leaving directory '/home/li/openwrt/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#38.97#8.29#28.21
make[3]: Entering directory '/home/li/openwrt/package/feeds/luci-app-passwall'
bash: % 86400: syntax error: operand expected (error token is "% 86400")
bash: % 86400: syntax error: operand expected (error token is "% 86400")
/home/li/openwrt/feeds/luci/luci.mk:277: *** Package/luci-app-passwall is missing the VERSION field.  Stop.
make[3]: Leaving directory '/home/li/openwrt/package/feeds/luci-app-passwall'
time: package/feeds/luci-app-passwall/compile#1.53#0.16#3.66
make[2]: *** [package/Makefile:113: package/feeds/luci-app-passwall/compile] Error 2
make[2]: Leaving directory '/home/li/openwrt'
make[1]: *** [package/Makefile:107: /home/li/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/li/openwrt'
make: *** [/home/li/openwrt/include/toplevel.mk:220：world] 错误 2

试了N次，上一次是luci-app-advancedsetting 这个插件编译不过去，现在是passwall。
我把PASSWALL去掉看看

garypang13, 2020-04-10:
openwrt官方源码调整导致的错误,等我后续修复下.

garypang13, 2020-04-11:
已修复.

lx521603, 2020-04-11:
Applying ./patches/fullconenat.patch using plaintext:
patching file defaults.c
Hunk #1 succeeded at 48 (offset 2 lines).
patching file options.h
Hunk #1 succeeded at 297 with fuzz 1 (offset 30 lines).
patching file zones.c

Applying ./patches/fullconenat.patch.1 using plaintext:
patching file defaults.c
Hunk #1 FAILED at 46.
1 out of 1 hunk FAILED -- saving rejects to file defaults.c.rej
patching file options.h
Hunk #1 FAILED at 267.
1 out of 1 hunk FAILED -- saving rejects to file options.h.rej
patching file zones.c
Hunk #1 FAILED at 627.
Hunk #2 FAILED at 712.
2 out of 2 hunks FAILED -- saving rejects to file zones.c.rej
Patch failed!  Please fix ./patches/fullconenat.patch.1

本地提示这个，云编译好像也是。我的编译不出来。

garypang13, 2020-04-11:
因为你本地执行了两次diy.sh 导致下载了fullconenat.patch和fullconenat.patch.1,删除package/network/config/firewall/patches/fullconenat.patch1即可.

lx521603, 2020-04-11:
不行，我编译不过去。

2小时前，我重新fork，然后改了点配置，云编译。

调整固件大小 ，因为我用到的比较少，我平时编译就只用到16,160
CONFIG_TARGET_KERNEL_PARTSIZE=16
CONFIG_TARGET_ROOTFS_PARTSIZE=220

# Applications    =y的大部分被我注释了，因为用不到。

CONFIG_PACKAGE_autocore=y         我改成了Y。

garypang13, 2020-04-11:
不好意思,又调整了下,再试试.
CONFIG_PACKAGE_autocore 不要开启,不兼容luci 19.07

garypang13, 2020-04-11:
我云编译通过了.

lx521603, 2020-04-11:
> 我云编译通过了.

我的又在1小时40多分的时候失败了，我现在把容量改回去了。（之前我编译同样的插件，都是16,160，都是成功的，）

我编译的插件有ssrp+   passwall   时间控制，oaf，多拨，负载平衡，adg , smartdns,adbyb+,ttyd,微软自动激活，minidlna，upnp。其它的什么都没有。
如果再失败，我就去掉oaf看看，因为这个插件，我本地编译是从来没成功过，去掉后都是成功的。

然后我改了默认ip地址。

Applications 部分，注释我不要的，希望能成功

lx521603, 2020-04-11:
我之前就在本地编译过官方master的， luci会出问题。

然后luci用lean大，lionel大的，都会出现lean大的插件好多都不兼容，编译成功，luci里也不会运行。

lionel的因为他那里也有官方master版，所以他的包里的插件一般都还是能运行。

我一般都是本地编译lionel的官方master版本。

本地编译纯官方master版本，加别人插件之类的，其实有个很大的问题，就是git pull的时候。所以你这个是最好的方法，希望能搞好。

lx521603, 2020-04-12:
pull今天最新的，终于编译成功了，插件试了下，暂时正常，有什么问题后续会反馈。

再次感谢！

closed 2020-04-21 by garypang13