kiddin9/Kwrt #6: 老大，拖到本地进行编译，每次都需要按照你写的来一遍吗？

sarsmark-git, 2020-04-13:
拖到本地进行编译，每次都需要按照你写的来一遍吗？
如果不需要的话，我如何才能同步你的更新。
能否麻烦再补充一下二次以后编译的教程，以便我等小白依葫芦画瓢。
谢谢。

garypang13, 2020-04-13:
二次编译执行脚本
```
cd openwrt && git pull && cd -
cd Actions-OpenWrt-Nginx && git pull && cd -
cp -Rf Actions-OpenWrt-Nginx/* openwrt/
cd openwrt
./scripts/feeds update -a && ./scripts/feeds install -a
./diy.sh
cp -Rf diy/* ./
mv X86_64.config .config
make defconfig
```

sarsmark-git, 2020-04-13:
多谢老大！！！

sarsmark-git, 2020-04-13:
mv X86_64.config .config这步可以不要吧，否则把我自己选好的又覆盖了

garypang13, 2020-04-13:
> mv X86_64.config .config这步可以不要吧，否则把我自己选好的又覆盖了

是的.

sarsmark-git, 2020-04-13:
执行./diy.sh，出现好多错误

比如：

atal: 目标路径 'files/usr/share/amule/webserver/AmuleWebUI-Reloaded' 已经存在，并且不是一个空目录。
fatal: 目标路径 'files/usr/share/aria2' 已经存在，并且不是一个空目录。
fatal: 目标路径 'package/lean/antileech/src' 已经存在，并且不是一个空目录。
取出版本 2450。
fatal: 目标路径 'luci-app-adguardhome' 已经存在，并且不是一个空目录。
fatal: 目标路径 'luci-theme-argon' 已经存在，并且不是一个空目录。
fatal: 目标路径 'luci-app-smartdns' 已经存在，并且不是一个空目录。
fatal: 目标路径 'luci-app-diskman' 已经存在，并且不是一个空目录。
mkdir: 无法创建目录 “parted”: 文件已存在
fatal: 目标路径 'luci-app-serverchan' 已经存在，并且不是一个空目录。
取出版本 108。
取出版本 69。
fatal: 目标路径 'OpenAppFilter' 已经存在，并且不是一个空目录。
取出版本 20。
取出版本 1797。
取出版本 1797。
取出版本 1797。
取出版本 1797。
fatal: 目标路径 'luci-app-dockerman' 已经存在，并且不是一个空目录。
fatal: 目标路径 'openwrt-adguardhome' 已经存在，并且不是一个空目录。
fatal: 目标路径 'luci-app-php-kodexplorer' 已经存在，并且不是一个空目录。
fatal: 目标路径 'luci-app-eqos' 已经存在，并且不是一个空目录

garypang13, 2020-04-13:
已修复.请再全新编译一次.

closed 2020-04-21 by garypang13