kiddin9/Kwrt #7: AdguardHome搭配SmartDNS,smartdns需要如何配置

dhlsam1987, 2020-04-13:
大佬，固件使用后，adguardhome好想配置文件都配置好了dns和去广告条目，如何搭配smartdns设置，启用第一第二DNS服务器后还需要设置什么么

garypang13, 2020-04-13:
smartdns也配置好了的,不用启动第二DNS服务器,把TCP服务器和域名预加载勾上就好了.

dhlsam1987, 2020-04-13:
设置成功了，这个根it奶爸那个视频原理一样，smartdns是adguard上游 然后adguard是dnsmasq上游

garypang13, 2020-04-13:
是的哈.

dhlsam1987, 2020-04-13:
大佬，非常好的固件，可以加上l大最新的airplay2和PCHiFi 数字转盘遥控还有新的cifs-mount这个插件么，这样软路由插了usb声卡就可以当播放器了，cifs-mount这个就不用老的smb 就可以挂载nas上的盘做下载盘了，我自己修改了config文件，可是每次云编译都编译失败
CONFIG_PACKAGE_luci-app-cifs-mount=y
CONFIG_PACKAGE_luci-app-music-remote-center=y
CONFIG_PACKAGE_luci-app-airplay2=y
airplay2需要的kernel 
CONFIG_PACKAGE_kmod-sound-core=y
CONFIG_PACKAGE_kmod-usb-audio=y

garypang13, 2020-04-13:
已加入cifs-mount和music-remote-center
airplay2和music-remote-center功能有重复么,一直没在用苹果设备,不太熟悉.

sarsmark-git, 2020-04-14:
openlash能加入吗，似乎不错的样子

sarsmark-git, 2020-04-14:
还有AdguardHome是默认启用的，smartdns又没有，这是故意的吗？能否都默认不启用。

garypang13, 2020-04-14:
更新到今天编译的版本,smartdns就默认开启了,
现在的这两个翻墙工具完全够用了,openlash相对更复杂,属于比较折腾的.后续我看需要的人多不多再看是否加入.

garypang13, 2020-04-14:
添加了PCHiFi 但是开启不了,也不报任何错误,你刷最新版测试下看怎么使用,不行就只能先去掉 @dhlsam1987

dhlsam1987, 2020-04-14:
airplay2和music-remote-center功能是不同的 ，一个是接收，一个是媒体播放服务器

garypang13, 2020-04-14:
airplay2需求太小众了 暂时先不加了.

dhlsam1987, 2020-04-14:
好嘞 感谢

imjunfeng, 2020-04-15:
airplay2加进去呗，挺好用的，老的音响接个小路由插个声卡，苹果设备就能投送无损音频了

garypang13, 2020-04-17:
> airplay2加进去呗，挺好用的，老的音响接个小路由插个声卡，苹果设备就能投送无损音频了

airplay2 暂时以独立插件的形式提供,可自行安装

closed 2020-04-21 by garypang13