kiddin9/Kwrt #22: 大佬，upnp好像用不了

DCLangX, 2020-04-30:
也不知道为啥，勾选输出日志，昨天一直报
`714 NoSuchEntryInArray`
今天改报这样
`Thu Apr 30 22:38:33 2020 daemon.debug miniupnpd[28457]: level=0 type=8
Thu Apr 30 22:38:33 2020 daemon.debug miniupnpd[28457]: ifindex = 6  192.168.233.1
Thu Apr 30 22:38:33 2020 daemon.debug miniupnpd[28457]: ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1 (ver=1)
Thu Apr 30 22:38:33 2020 daemon.info miniupnpd[28457]: SSDP M-SEARCH from 192.168.233.238:51864 ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1
Thu Apr 30 22:38:33 2020 daemon.info miniupnpd[28457]: Single search found
Thu Apr 30 22:38:33 2020 daemon.debug miniupnpd[28457]: SendSSDPResponse(): 0 bytes to 192.168.233.238:51864 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1
 USN: uuid:6cb1d8e6-bf51-4277-8055-5c641c08fd60::urn:schemas-upnp-org:device:InternetGatewayDevice:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20191006
 LOCATION: http://192.168.233.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1588177482
 BOOTID.UPNP.ORG: 1588177482
 CONFIGID.UPNP.ORG: 1337
Thu Apr 30 22:38:36 2020 daemon.debug miniupnpd[28457]: level=0 type=8
Thu Apr 30 22:38:36 2020 daemon.debug miniupnpd[28457]: ifindex = 6  192.168.233.1
Thu Apr 30 22:38:36 2020 daemon.debug miniupnpd[28457]: ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1 (ver=1)
Thu Apr 30 22:38:36 2020 daemon.info miniupnpd[28457]: SSDP M-SEARCH from 192.168.233.238:51864 ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1
Thu Apr 30 22:38:36 2020 daemon.info miniupnpd[28457]: Single search found
Thu Apr 30 22:38:36 2020 daemon.debug miniupnpd[28457]: SendSSDPResponse(): 0 bytes to 192.168.233.238:51864 ST: HTTP/1.1 200 OK
 CACHE-CONTROL: max-age=120
 ST: urn:schemas-upnp-org:device:InternetGatewayDevice:1
 USN: uuid:6cb1d8e6-bf51-4277-8055-5c641c08fd60::urn:schemas-upnp-org:device:InternetGatewayDevice:1
 EXT:
 SERVER: OpenWRT/SNAPSHOT UPnP/1.1 MiniUPnPd/2.1.20191006
 LOCATION: http://192.168.233.1:5000/rootDesc.xml
 OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
 01-NLS: 1588177482
 BOOTID.UPNP.ORG: 1588177482
 CONFIGID.UPNP.ORG: 1337`

garypang13, 2020-04-30:
[image]

我这边没问题,你下载个迅雷打开,看会不会自动添加端口转发.

DCLangX, 2020-05-01:
折腾了一下，群晖可以映射了，迅雷好像还不行，一会儿再试试看

closed 2020-05-01 by DCLangX