kiddin9/Kwrt #29: ShadowsocksR Plus移除了

lexin8, 2020-05-05:
passwall原作者已经弃更了，个人觉得不好用，换节点延迟大，而且经常运行不起来 ，需要重启插件。
SSR最近测试发现DNS有问题，会变成全局模式，可能是跟smartdns、AdGuardHome有关系。

garypang13, 2020-05-05:
我没在用passwall, bug多,性能不行,还弃更了,下版考虑去掉, SSR已经够好用了,
SSR如果使用gfwlist模式那AdGuardHome就不要使用重定向53端口,而是使用作为dnsmasq的上游服务器,不然就会变成全局模式

lexin8, 2020-05-06:
这两天都在编译你的固件，主题不错，可是AdGuardHome、smartdns、猫咪三剑客真的不建议加进去。这两天的版本有问题rm -rf .config后默认X86的自己选会报错，使用你的X86config设置在添加其他插件的话也报错，Docker选项正常了可是编出来运行不了。个人建议去除passwall 优化添加SSR就好，三剑客没觉得有多大用，去广告很鸡肋的功能就没开过。

lexin8, 2020-05-06:
还有一个共享工具，一开始我以为是Samba之后看插件才知道不是，很难用，无法写入也不知道怎么加密。建议优化Samba共享添加root用户密码也是root，这个添加用户加密功能需求比较大，我都是手动设置。

garypang13, 2020-05-06:
AdGuardHome、smartdns我自己在用,挺好用的,
Clash并没有编译进去,而是以插件形式提供
.config个别插件有warning警告并不影响编译和使用的
那个网络共享就是内核级Samba,和Samba配置和使用方法一模一样,我就是用的这个做文件共享, https://github.com/coolsnowwolf/lede/issues/3135

lexin8, 2020-05-06:
刚刚又跑了一遍，Docker运行不了 ，应该是缺少组件。我是双WAN口一条外网一条IPTV内外 运行AdGuardHome smartdns 的话会上不了网，就不敢开启。

garypang13, 2020-05-06:
[image]
我的Docker没问题,你是自己编译的固件么,
我用的是combined-efi.img.gz,每次编译完的固件都是在后台升级的.

lexin8, 2020-05-06:
这几天更新脚本就这样了，之前的可以
[image]
[image]

garypang13, 2020-05-06:
没启动的情况下是不显示docker相关信息的,直接勾选启动.运行起来后就能看到相关信息.

ledewrt, 2020-05-07:
adgaurdhome 无法过滤广告，同样的规则和配置在19。07和18系列上没有问题。smartdns一直显示没有运行。

garypang13, 2020-05-07:
> adgaurdhome 无法过滤广告，同样的规则和配置在19。07和18系列上没有问题。smartdns一直显示没有运行。

无法过滤广告去adg后台查询日志中确认是否所有DNS查询都有经过adg
smartdns一直显示没有运行是因为smartdns设置中默认勾选了IPV6,这个问题今天修复了.

lexin8, 2020-05-08:
二次编译要删除干净才能跑成功，相当于重新来一遍要好几个小时，为何不做成源代码不用脚本方式呢
[image]

garypang13, 2020-05-09:
做成源代码当然简单多了,但是你改过的文件就没法自动更新了,

二次编译脚本
```
rm -Rf Actions-OpenWrt-Nginx && git clone https://github.com/garypang13/Actions-OpenWrt-Nginx
cp -Rf Actions-OpenWrt-Nginx/* openwrt/
cd openwrt
git pull
rm -Rf feeds
./scripts/feeds update -a
./diy.sh
make defconfig
```

lexin8, 2020-05-09:
二次编译应该在加个命令mv X86_64.config .config 
都做成这样了，二次编译应该搞个一键脚本安装好编译环境

garypang13, 2020-05-09:
加上 mv X86_64.config .config 就把以前自己选的.config覆盖了,
编译环境在第一次编译的时候就弄好了,

closed 2020-05-09 by garypang13

lexin8, 2020-05-09:
https://github.com/lexin8/Actions-OpenWrt-Nginx/runs/658423942?check_suite_focus=true

The mentality exploded

garypang13, 2020-05-09:
官方仓库中没有samba了,只有samba4,