kiddin9/Kwrt #33: 本地编译的警告，影响编译成功或使用吗

yangxiangguo, 2020-05-09:
WARNING: Makefile 'package/feeds/luci/luci-app-e2guardian/Makefile' has a dependency on 'e2guardian', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-samba/Makefile' has a dependency on 'samba36-server', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-e2guardian/Makefile' has a dependency on 'e2guardian', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-passwall/Makefile' has a dependency on 'brook', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-samba/Makefile' has a dependency on 'samba36-server', which does not exist
WARNING: Makefile 'package/feeds/custom/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist

garypang13, 2020-05-09:
不影响的.

yangxiangguo, 2020-05-09:
error: patch failed: package/network/services/dnsmasq/files/dnsmasq.init:868
error: package/network/services/dnsmasq/files/dnsmasq.init: patch does not apply
error: patch failed: feeds/luci/modules/luci-base/root/usr/share/rpcd/acl.d/luci-base.json:63
error: feeds/luci/modules/luci-base/root/usr/share/rpcd/acl.d/luci-base.json: patch does not apply
error: patch failed: feeds/luci/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/10_system.js:15
error: feeds/luci/modules/luci-mod-status/htdocs/luci-static/resources/view/status/include/10_system.js: patch does not apply
error: patch failed: feeds/luci/applications/luci-app-transmission/luasrc/controller/transmission.lua:9
error: feeds/luci/applications/luci-app-transmission/luasrc/controller/transmission.lua: patch does not apply
error: patch failed: package/network/services/dnsmasq/files/dnsmasq.init:898
error: package/network/services/dnsmasq/files/dnsmasq.init: patch does not apply
error: patch failed: feeds/packages/net/aria2/files/aria2.init:312
error: feeds/packages/net/aria2/files/aria2.init: patch does not apply
error: feeds/packages/netdata/Makefile: No such file or directory
error: feeds/packages/netdata/files/netdata.init: No such file or directory
error: patch failed: feeds/luci/applications/luci-app-firewall/htdocs/luci-static/resources/view/firewall/zones.js:45
error: feeds/luci/applications/luci-app-firewall/htdocs/luci-static/resources/view/firewall/zones.js: patch does not apply

这种的影响吗 执行diy.sh报的

garypang13, 2020-05-10:
这个是多次执行./diy.sh导致的,不影响.

yangxiangguo, 2020-05-10:
编译已成功，感谢大佬~

closed 2020-05-10 by garypang13