kiddin9/Kwrt #39: opentomato的themes出错

hong0980, 2020-05-16:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_status/index'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/opentomato/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:755: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:755: in function 'include'
	[string "/usr/lib/lua/luci/view/admin_status/index.h..."]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:1015: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:979: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

garypang13, 2020-05-16:
感谢反馈,  下一版会修复.

garypang13, 2020-05-19:
已修复.

closed 2020-05-19 by garypang13