kiddin9/Kwrt #44: 域名劫持IP端口

lexin8, 2020-05-18:
请教下大神，使用hosts劫持只能是域名，实现劫持1.cn域名到192.168.1.1:8888这样带端口怎么实现呢？

garypang13, 2020-05-18:
先将1.cn DNS重写到路由器IP
然后在nginx中添加1.cn的301重定向到192.168.1.1:8888,或者添加1.cn的反向代理到192.168.1.1:8888

garypang13, 2020-05-18:
https://github.com/garypang13/Actions-OpenWrt-Nginx/blob/master/files/etc/nginx/conf.d/reverse_proxy.sample
添加了个反向代理的示例,

lexin8, 2020-05-18:
是哪里错了？小白不太懂
hosts做域名劫持
nginx.conf文件下添加

server
    {
        listen 80;
        server_name 1.cn;
		
        location / {
		proxy_pass              http://192.168.1.1:8888;
		proxy_http_version      1.1;
		proxy_set_header        X-Forwarded-Host        $server_name:$server_port;
		proxy_hide_header       Referer;
		proxy_hide_header       Origin;
		proxy_set_header        Referer                 '';
		proxy_set_header        Origin                  '';
}
    }

garypang13, 2020-05-18:
不要加在nginx.conf中
在 /etc/nginx/conf.d/文件夹中创建 1.cn.conf 文件
加到 1.cn.conf这个文件中

lexin8, 2020-05-18:
/etc/nginx/没有conf.d 直接添加1.cn.conf 文件，添加代码。
反反复复调试还是没成功，也网上找了教程还是一无所获。

garypang13, 2020-05-18:
你是用的我的固件么,
[image]

lexin8, 2020-05-19:
[image]
默认是没有的我在这里加上，还需要选其他配置吗

garypang13, 2020-05-19:
不是用的我的固件的话,如果你不懂nginx的话配置比较麻烦,

closed 2020-05-19 by garypang13

lexin8, 2020-05-19:
这个就是你的源码，今天搞几个云编译都失败，现在重新fork重新跑一下。