kiddin9/Kwrt #50: 编译几次，都是最后报错。

jarod360, 2020-05-25:
cc1: some warnings being treated as errors
scripts/Makefile.build:265: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.o' failed
make[5]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.o] Error 1
make[4]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
Makefile:1700: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[3]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[4]: Leaving directory '/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.42'
Makefile:41: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/feeds/custom/luci/oaf'
time: package/feeds/custom/oaf/compile#1.47#0.37#1.85
package/Makefile:111: recipe for target 'package/feeds/custom/oaf/compile' failed

garypang13, 2020-05-25:
https://github.com/destan19/OpenAppFilter/issues/49
luci-app-oaf这个软件的问题,建议先去掉.

jarod360, 2020-05-25:
谢谢大佬指点

closed 2020-05-26 by garypang13