kiddin9/Kwrt #51: 老大编译成功了，打开首页报错。

jarod360, 2020-05-25:
/usr/lib/lua/luci/model/cbi/passwall/api/v2ray.lua:6: module 'luci.model.ipkg' not found:
	no field package.preload['luci.model.ipkg']
	no file './luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg/init.lua'
	no file '/usr/lib/lua/luci/model/ipkg.lua'
	no file '/usr/lib/lua/luci/model/ipkg/init.lua'
	no file './luci/model/ipkg.so'
	no file '/usr/lib/lua/luci/model/ipkg.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/passwall/api/v2ray.lua:6: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/passwall.lua:8: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1076: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:1187: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:694: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:795: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

garypang13, 2020-05-25:
应该是你修改.config文件的时候去掉了某个不该去掉的.
命令行输入
opkg update
opkg install luci-lib-ipkg
可解决module 'luci.model.ipkg' not found:的问题

jarod360, 2020-05-25:
好的，我试下

jarod360, 2020-05-25:
> 应该是你修改.config文件的时候去掉了某个不该去掉的.
> 命令行输入
> opkg update
> opkg install luci-lib-ipkg
> 可解决module 'luci.model.ipkg' not found:的问题
报错的问题解决了。查看了下x86_64.config 里面默认没有安装这个 luci-lib-ipkg
。

jarod360, 2020-05-25:
还有个问题 就是x86_64.config里默认设置安装dockerman和docker-ce 但是进入系统发现，docker页面不显示系统版本号相关讯息，卸载docker-ce后。重新安装。提示很多依赖没有安装导致docker不对。
containerd (31.1 MB) 未安装
runc (3.1 MB) 未安装
libdevmapper (147.8 KB) 未安装
kmod-dm 已安装
kmod-crypto-manager 已安装
kmod-crypto-aead 已安装
kmod-crypto-null 已安装
kmod-crypto-pcompress 已安装
kmod-dax 已安装
libnetwork (8.2 MB) 未安装
tini (28.9 KB) 未安装
libwebsockets-full 未安装
luci-lib-docker.ipk 未安装
之前用您的原始文件编译加入docker也是同样缺少相关依赖，导致docker无法使用。您可以排查下

garypang13, 2020-05-25:
docker的系统版本号相关讯息要启动docker后才显示的.
[image]

jarod360, 2020-05-25:
> docker的系统版本号相关讯息要启动docker后才显示的.
> [image]
可是我默认至显示概览一项，其他的都没有显示。除非把相关依赖全装了。不知道不是因为没有安装VPN和网盘的软件导致相关依赖没有安装。用您的原始代码编译完。发现也是只显示概览一向。

jarod360, 2020-05-25:
默认只装了截图中的软件，其他不用的全去掉了。不知道是不是因为这个原因导致的。
[image]

garypang13, 2020-05-25:
docker的其他菜单也要启动docker后才会显示.你现在把docker依赖都卸载掉了,需要重刷一遍固件才行

garypang13, 2020-05-25:
luci-lib-ipkg已加上,很多插件都会自动添加luci-lib-ipkg依赖所以没注意,

jarod360, 2020-05-25:
您的意思是，默认docker没有启动是吗？是需要通过命令行启动才可以是吗

garypang13, 2020-05-25:
在后台docker设置中勾选启动,保存并应用即可.docker会在几秒后启动完成.

closed 2020-05-27 by garypang13