kiddin9/Kwrt #53: 直接Fork，什么都没改  云编译报错

ng0212, 2020-05-25:
x86_64-openwrt-linux-musl-gcc -I. -I./ -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DZLIB_CONST -DHAVE_AV_CONFIG_H -DBUILDING_avformat -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-4.2.2=ffmpeg-4.2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -fpic    -std=c11 -fomit-frame-pointer -fPIC -pthread -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -Wdeclaration-after-statement -Wall -Wdisabled-optimization -Wpointer-arith -Wredundant-decls -Wwrite-strings -Wtype-limits -Wundef -Wmissing-prototypes -Wno-pointer-to-int-cast -Wstrict-prototypes -Wempty-body -Wno-parentheses -Wno-switch -Wno-format-zero-length -Wno-pointer-sign -Wno-unused-const-variable -Wno-bool-operation -Wno-char-subscripts -O3 -fno-math-errno -fno-signed-zeros -flto -fno-tree-vectorize -Werror=format-security -Werror=implicit-function-declaration -Werror=missing-prototypes -Werror=return-type -Werror=vla -Wformat -fdiagnostics-color=auto -Wno-maybe-uninitialized  -MMD -MF libavformat/mpegenc.d -MT libavformat/mpegenc.o -c -o libavformat/mpegenc.o libavformat/mpegenc.c
x86_64-openwrt-linux-musl-gcc -I. -I./ -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -DZLIB_CONST -DHAVE_AV_CONFIG_H -DBUILDING_avformat -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/ffmpeg-full/ffmpeg-4.2.2=ffmpeg-4.2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -fpic    -std=c11 -fomit-frame-pointer -fPIC -pthread -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWr

garypang13, 2020-05-25:
这个fork的已经是三天前的了.

ng0212, 2020-05-26:
> 这个fork的已经是三天前的了.

从上个星期开始每天fork一次，每次都卡编译

closed 2020-05-26 by ng0212