kiddin9/Kwrt #54: 希望大佬增加KoolProxyR plus+插件

jarod360, 2020-05-26:
这个广告过滤的好像挺彻底的。可以作为备选。

jarod360, 2020-05-26:
反馈个问题，使用内置docker ，创建网络模式为macvlan时，网络接口中docker创建的接口显示
协议: 静态地址
接收: 0 B (0 数据包)
发送: 0 B (0 数据包)
错误: 未知错误（DEVICE_CLAIM_FAILED）
容器也无法使用此接口。其他人的固件测试创建的接口显示正常。希望大佬能排查下。

garypang13, 2020-05-26:
koolshare koolproxyR很好用,但是移植到openwrt的版本功能不全,bug很多,单独作为插件包形式提供吧.

garypang13, 2020-05-26:
> 反馈个问题，使用内置docker ，创建网络模式为macvlan时，网络接口中docker创建的接口显示
> 协议: 静态地址
> 接收: 0 B (0 数据包)
> 发送: 0 B (0 数据包)
> 错误: 未知错误（DEVICE_CLAIM_FAILED）
> 容器也无法使用此接口。其他人的固件测试创建的接口显示正常。希望大佬能排查下。

确实存在,我去dockerman反馈下.

garypang13, 2020-05-26:
> > 反馈个问题，使用内置docker ，创建网络模式为macvlan时，网络接口中docker创建的接口显示
> > 协议: 静态地址
> > 接收: 0 B (0 数据包)
> > 发送: 0 B (0 数据包)
> > 错误: 未知错误（DEVICE_CLAIM_FAILED）
> > 容器也无法使用此接口。其他人的固件测试创建的接口显示正常。希望大佬能排查下。
> 
> 确实存在,我去dockerman反馈下.

还需要安装 kmod-macvlan 下一版会默认添加上

jarod360, 2020-05-26:
谢谢大佬。

jarod360, 2020-05-26:
不好意思，又麻烦大佬。希望大佬再编译的时候增加内核版本号官方源读取和回写功能的脚本，保持和官方源的内核版本号一致。这样某些官方源的核心模块就可以安装了。

jarod360, 2020-05-26:
老大抱歉，再次打扰，配置文件怎么设置才可以编译成内核版本是4.19的。使用5.4.4.2的内核。日志里docker提示很多与内核版本不兼容的消息。把CONFIG_PACKAGE_kmod-fs-exfat=y #Will only be built for kernel 4.19 设置成了这样，编译出来还是5.4.2的内核。

garypang13, 2020-05-26:
> 不好意思，又麻烦大佬。希望大佬再编译的时候增加内核版本号官方源读取和回写功能的脚本，保持和官方源的内核版本号一致。这样某些官方源的核心模块就可以安装了。

可以通过命令行安装 opkg install xxx --force-depends 忽略依赖

garypang13, 2020-05-26:
> 老大抱歉，再次打扰，配置文件怎么设置才可以编译成内核版本是4.19的。使用5.4.4.2的内核。日志里docker提示很多与内核版本不兼容的消息。把CONFIG_PACKAGE_kmod-fs-exfat=y #Will only be built for kernel 4.19 设置成了这样，编译出来还是5.4.2的内核。

需要修改 openwrt/target/linux/x86/Makefile 中的 KERNEL_PATCHVER:=5.4 为4.19

closed 2020-05-26 by jarod360