kiddin9/Kwrt #57: upnp 功能无法使用

goofysun, 2020-05-27:
fork 之后 尝试 通过替换 2.1.1 版本的upnp 来解决，但是已知编译失败。

@garypang13 的upnp 功能是有效的么

garypang13, 2020-05-27:
我在电脑上打开迅雷,能自动添加端口转发.

[image]

goofysun, 2020-05-27:
> 我在电脑上打开迅雷,能自动添加端口转发.
> 
> [image]

感谢回复，
我的不行，不过我路由器 wan 口是 dhcp 获取ip地址的，光猫没改桥接，目前看下来就这点不一样了，不知道会不会有影响

DCLangX, 2020-05-29:
这块比较玄学，我起初用的月初的版本，一开始开启upnp后，我的群晖始终无法找到upnp设备，迅雷也不行，然后我参考这个改了下upnp设置，https://koolshare.cn/thread-157946-1-1.html
改之后群晖认得到upnp设备了，但是应用upnp规则时又不行，然后我就勾选了启用igdv1，果不其然又识别不到upnp设备了，但是重启群晖后，我却发现自动映射规则成功了，尽管识别不到upnp设备。。。满心欢喜然后试了下迅雷，依然还是不行，无论怎么重启都没用，有点遗憾。

上个礼拜，我又编译了一个版本，发现upnp彻底坏了....再怎么折腾也没用，群晖也凉凉，心灰意冷于是我又换回了旧版本。。然后震惊的发现都能用了，明明之前怎么重启都不行，也没有重装，因为用的是虚拟机，根本没动过，迅雷、yy都ok，群晖也完美，现在有点不敢重启。。。怕又玄学失效了

lx521603, 2020-05-29:
楼上的几个。
“启用 IGDv1 模式”打勾试试，已知的是迅雷和其它BT软件，用的模式是不一样的。如果其他软件upnp成功了，迅雷就不会成功，反过来也一样。反正不能所有软件都upnp成功，除非用其它作为主路由，op作为旁路由。OP作为主路由就是这样的。

所以，迅雷不成功，开其它软件试试。或者迅雷成功了，看看其它软件会不会成功。

我没有测试，我的OP不是主路由，这个东西是L大说的。

goofysun, 2020-05-30:
我这比较玄学。

刚启动的时候会映射成功几个端口，然后后面就不行了。

我 也有一次启动 upnp 端口映射都正常，但是重启一次之后就跪了。

现在 找了个 老版本内核固件，用upnp 2.0.2 …… 一点问题没有

goofysun, 2020-05-30:
哎，老版本的内核虽然能映射，也是一堆问题，重复映射导致 nas 自动换端口了，映射了一堆端口。

还是做旁路由吧

lx521603, 2020-05-30:
> 哎，老版本的内核虽然能映射，也是一堆问题，重复映射导致 nas 自动换端口了，映射了一堆端口。
> 
> 还是做旁路由吧

有些是软件的问题，我的transmission貌似从来没成功过，qbittorrent好像只有启动的时候会检查，但deluge永远都会成功（无论爱快，op，高恪，ROS都这样）。

garypang13, 2020-05-30:
这个问题真的没有完美的解决方案,我反正是尽量手动添加映射.

goofysun, 2020-05-30:
确实不好解决，openwrt 的upnp 太迷糊。
一些固定端口手动还行，下载软件的很多 是随机的，不支持upnp 就没有上传

closed 2020-06-02 by garypang13

liuran001, 2020-08-05:
@garypang13 有一个较完美的解决方案，就是换2017版的miniupnpd，对国内软件比较友好，在lean源码里面有  
我现在就是这样，找了很多人测试，都还不错

DCLangX, 2020-08-05:
> @garypang13 有一个较完美的解决方案，就是换2017版的miniupnpd，对国内软件比较友好，在lean源码里面有
> 我现在就是这样，找了很多人测试，都还不错

我也看到那个issue了，谁能想到miniupnp必须要wan口是公网ip才能生效。。。离谱
我是光猫拨号，光猫开dmz给软路由，所以wan口当然是内网的ip。。。
后来照里面大佬所说，关了ipv6，然后配置里加入
`option external_ip '随意一个公网IP'`
就可以正常转发了
coolsnowwolf/lede#5020

liuran001, 2020-08-05:
> > @garypang13 有一个较完美的解决方案，就是换2017版的miniupnpd，对国内软件比较友好，在lean源码里面有
> > 我现在就是这样，找了很多人测试，都还不错
> 
> 我也看到那个issue了，谁能想到miniupnpd必须要wan口是公网ip才能生效。。。离谱
> 我是光猫拨号，光猫开dmz给软路由，所以wan口当然是内网的ip。。。
> 后来照里面大佬所说，关了ipv6，然后配置里加入
> `option external_ip '随意一个公网IP'`
> 就可以正常转发了
> [coolsnowwolf/lede#5020](https://github.com/coolsnowwolf/lede/issues/5020)

@DCLangX 后面不是有人说是心理作用吗emmmmm'

DCLangX, 2020-08-05:
> @DCLangX 后面不是有人说是心理作用吗emmmmm'

他是在回复那个**处于限制型NAT**的兄弟，这样做虽然他upnp能强制开启，但是他软路由本身在内网里，已经被nat了一次，所以开了也没用，不过我这种情况不一样，我虽然软路由也在内网，公网ip在光猫上，但是光猫直接dmz穿透给软路由，等于软路由直接暴露再公网，所以不是心理作用，只要upnp不掉链子能强制开启就能使用

liuran001, 2020-08-05:
> 他是在回复那个**处于限制型NAT**的兄弟，这样做虽然他upnp能强制开启，但是他软路由本身在内网里，已经被nat了一次，所以开了也没用，不过我这种情况不一样，我虽然软路由也在内网，公网ip在光猫上，但是光猫直接dmz穿透给软路由，等于软路由直接暴露再公网，所以不是心理作用，只要upnp不掉链子能强制开启就能使用

但是手动设置还是更麻烦一些，17版就一切正常，lean源码现在就是17

garypang13, 2020-08-05:
@liuran001 以更换为17版,