kiddin9/Kwrt #62: 网络异常

lexin8, 2020-06-05:
问题1：固件默认设置状态SSR开启会变成全局
问题2：去广告测试爱奇艺腾讯电脑端过滤不了，只有手机爱奇艺可以过滤。
问题3：关闭AdGuard Home以及SmartDNS 后SSR改为PDNSD模式才不会被全局，可是打开油管视频偶尔播放不出来。电脑打开网页很慢尤其是百度首页，PING 爱奇艺掉PING，ping百度正常。

[image]
[image]

[image]

lexin8, 2020-06-05:
GFW模式大陆模式都测试，均有问题，总结：DNS负优化了或者不适用其他网络环境。

garypang13, 2020-06-05:
应该是你网络环境的问题,我的DNS优化只针对adguardhome和smartdns,

lexin8, 2020-06-05:
同样的环境固件不一样而已

garypang13, 2020-06-05:
最好直接用我编译好的固件测试下.看还有没问题. 
https://github.com/rufengsuixing/luci-app-adguardhome/issues/6#issuecomment-559812665 adguardhome主要还是过滤掉无用的DNS,加速网站访问.

lexin8, 2020-06-05:
重新编译别的源码固件网络都是秒开，你的源码我基本都是默认状态，没有添加第三方。今天卡死了，
在提几个BUG，手机登录管理页面菜单栏很难点出来
2.SSR订阅要点一下加号才能保存，更新订阅节点要先删除更新才有反应，研究了很久，而且会被全局问题困扰。
个人建议还是别搞DNS了默认就好，最多加个去爱奇艺广告就行 ，腾讯过滤不了。过滤广告我感觉是杀敌一千自损一千。
[image]

garypang13, 2020-06-05:
.SSR订阅要点一下加号才能保存,新增加的订阅当然要点加号才算添加成功的.过滤广告问题我已经解释过了,不只是过滤广告,而是能加快网站打开速度.目前并没发现任何副总用.
DNS问题我也看不到你自己编译的源码是否有修改过哪里,用我编译好的固件,全新安装才能排除所有可能的因素.

lexin8, 2020-06-05:
用别的固件不需要点加号填进去，正常保存，不能正常保存这问题很多人在别的固件上都有反映，源的问题。lede用的https://github.com/fw876/helloworld是正常的。这都是小问题，主要是被全局问题。

前面编译的固件反映后按照你要答复，不使用5335 SSR就正常了，今天是全部重新下载编译环境以及源码重新编译了2个来小时，在测试了一天。云编译只能用默认配置，添加插件会编译失败，很奇怪，无奈今天用本地环境编译的。测试了一天 关掉AD跟DNS2个插件SSR也关了测试还是卡。

garypang13, 2020-06-05:
卡的话会不会是软路由配置太低了,高版本内核对配置要求高些,
全局问题是特例,并未有其他人反映过此问题,不是用我编译好的固件,全新安装的话我也无法解决.

closed 2020-06-06 by garypang13