kiddin9/Kwrt #65: ipsec

4041110, 2020-06-08:
你好，请问怎么添加ipsec插件

garypang13, 2020-06-08:
ipsec插件依赖 https://github.com/coolsnowwolf/packages/tree/master/net/strongswan  
所以要再diy.sh中添加 svn co https://github.com/coolsnowwolf/packages/trunk/net/strongswan

4041110, 2020-06-09:
你好，请问插件依赖是在哪里添加呢

garypang13, 2020-06-09:
添加好了,将.config中的CONFIG_PACKAGE_luci-app-ipsec-vpnd=n改为CONFIG_PACKAGE_luci-app-ipsec-vpnd=y即可.

4041110, 2020-06-09:
好的，谢谢您

4041110, 2020-06-10:
> 添加好了，将.config中的CONFIG_PACKAGE_luci-app-ipsec-vpnd = n替换CONFIG_PACKAGE_luci-app-ipsec-vpnd = y即可。

你好，直接把n改为y后编译失败，麻烦帮我看下是什么问题可以吗
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..  -Wall -W  -DGRUB_MACHINE_EFI=1 -DGRUB_MACHINE=X86_64_EFI -m64 -nostdinc -isystem /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/include -I../include -I../include -DGRUB_FILE=\"video/readers/png.c\" -I. -I. -I.. -I.. -I../include -I../include -I../grub-core/lib/libgcrypt-grub/src/    -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -D_FILE_OFFSET_BITS=64 -Os -Wall -W -Wshadow -Wpointer-arith -Wundef -Wchar-subscripts -Wcomment -Wdeprecated-declarations -Wdisabled-optimization -Wdiv-by-zero -Wfloat-equal -Wformat-extra-args -Wformat-security -Wformat-y2k -Wimplicit -Wimplicit-function-declaration -Wimplicit-int -Wmain -Wmissing-braces -Wmissing-format-attribute -Wmultichar -Wparentheses -Wreturn-type -Wsequence-point -Wshadow -Wsign-compare -Wswitch -Wtrigraphs -Wunknown-pragmas -Wunused -Wunused-function -Wunused-label -Wunused-parameter -Wunused-value  -Wunused-variable -Wwrite-strings -Wnested-externs -Wstrict-prototypes -g -Wredundant-decls -Wmissing-prototypes -Wmissing-declarations  -Wextra -Wattributes -Wendif-labels -Winit-self -Wint-to-pointer-cast -Winvalid-pch -Wmissing-field-initializers -Wnonnull -Woverflow -Wvla -Wpointer-to-int-cast -Wstrict-aliasing -Wvariadic-macros -Wvolatile-register-var -Wp
This file has been truncated. View the full logs in the  menu once the check is completed.
 Upload bin directory

4041110, 2020-06-10:
请问一直编译失败是什么问题呢
4872916Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
2020-06-10T07:39:38.4873873Z make: *** [world] Error 2
2020-06-10T07:39:38.4941182Z ##[error]Process completed with exit code 2.
2020-06-10T07:39:38.5459515Z Post job cleanup.
2020-06-10T07:39:39.0022057Z [command]/usr/bin/git version
2020-06-10T07:39:39.0075043Z git version 2.27.0
2020-06-10T07:39:39.0128038Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-06-10T07:39:39.0180702Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-06-10T07:39:39.0562485Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-06-10T07:39:39.0607010Z http.https://github.com/.extraheader
2020-06-10T07:39:39.0617482Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-06-10T07:39:39.0665954Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-06-10T07:39:39.0979684Z Cleaning up orphan processes

garypang13, 2020-06-10:
已经修复了,更新到最新再试试.

4041110, 2020-06-10:
好的谢谢，我在试下

4041110, 2020-06-11:
直接修改后插件列表没有显示，请问还有别的地方要修改的吗

4041110, 2020-06-11:
麻烦抽空看一下可以吗，谢谢了，编译几次成功了但是路由器后台还是找不到这个

garypang13, 2020-06-11:
@4041110 已修复,更新后再试试.

4041110, 2020-06-11:
> @ 4041110已修复，更新后再试试。

好的谢谢

closed 2020-06-13 by 4041110