kiddin9/Kwrt #71: 插件收录提议

garypang13, 2020-06-13:
本固件内置了自建的插件仓库服务器,作为官方源的补充,收录了市面上大部分主流第三方插件,如在软件列表中没有找到自己想要的插件,欢迎在下面留言(在有源码的情况下)

本固件默认只包含最基础的插件,其他插件需要在后台软件包中自行安装,系统升级不会丢失已安装的插件.

CORE2021, 2020-06-21:
luci-app-vssr

garypang13, 2020-06-21:
> luci-app-vssr

已添加

CORE2021, 2020-06-22:
luci-app-dnscrypt-proxy2

CORE2021, 2020-06-22:
luci-theme-infinityfreedom
https://github.com/xiaoqingfengATGH/luci-theme-infinityfreedom

CORE2021, 2020-06-22:
安装好些软件包都是会报uci错误，虽然不影响后续使用
还有就是odhcpd好像获取不到前缀，我自己刷了验证过三次了

garypang13, 2020-06-22:
> luci-theme-infinityfreedom
> https://github.com/xiaoqingfengATGH/luci-theme-infinityfreedom

这主题没有适配luci19.07

garypang13, 2020-06-22:
> 安装好些软件包都是会报uci错误，虽然不影响后续使用
> 还有就是odhcpd好像获取不到前缀，我自己刷了验证过三次了
> 我是光猫和路由器是同一个LAN桥接，WAN6使用@lan接口（LAN别名）拨号

uci报错不用管
odhcpd获取不到前缀是啥意思呢,不好意思我没用过IPV6,v只是测试了能正确获取到IPV6数据
[image]

CORE2021, 2020-06-22:
我看系统日志会提示Permission Denied,很懵逼，IPV6的地址和前缀都没有，实在不知道哪里出了问题

garypang13, 2020-06-22:
> 我看系统日志会提示Permission Denied,很懵逼，IPV6的地址和前缀都没有，实在不知道哪里出了问题

开启IPV6需要修改 /etc/sysctl.d/99-custom.conf 删掉
```
net.ipv6.conf.lo.disable_ipv6=1
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1 
```

CORE2021, 2020-06-22:
> —这样吗，学到了

NeedforGit, 2020-07-15:
udp2raw

udpspeederv2

garypang13, 2020-07-15:
> udp2raw
> 
> udpspeederv2

有仓库地址么,

garypang13, 2020-07-15:
> udp2raw
> 
> udpspeederv2

已收录

Tiexcn, 2020-07-25:
https://github.com/destan19/OpenAppFilter
OpenAppFilter

garypang13, 2020-07-25:
@Tiexcn 这个插件没有适配luci19.07,暂时不收录

justin-eng, 2020-08-18:
kmod-usb-printer
安装不上， kernel不兼容

CoolZxp, 2020-09-10:
ssocks

CoolZxp, 2020-09-10:
passwall的依赖
ssocks

CoolZxp, 2020-09-10:
passwall的依赖
ssocks
[https://github.com/xiaorouji/openwrt-package/tree/master/package/ssocks](url)
[https://github.com/tostercx/ssocks](url)

garypang13, 2020-09-14:
> passwall的依赖
> ssocks
> [https://github.com/xiaorouji/openwrt-package/tree/master/package/ssocks](url)
> [https://github.com/tostercx/ssocks](url)

ssocks已经有了的.

fhleo, 2020-10-11:
京东签到
luci-app-jd-dailybonus
https://github.com/jerrykuku/luci-app-jd-dailybonus

yhl452493373, 2020-11-04:
大佬知道koolshare里面那个文件管理是那个插件吗？想要个那个功能...

garypang13, 2020-11-05:
> 大佬知道koolshare里面那个文件管理是那个插件吗？想要个那个功能...

没用过koolshare固件,你可以截图我看下.

yhl452493373, 2020-11-17:
> > 大佬知道koolshare里面那个文件管理是那个插件吗？想要个那个功能...
> 
> 没用过koolshare固件,你可以截图我看下.

用你最新的编出来有文件管理了，感谢

jackadam1981, 2020-11-30:
https://blog.csdn.net/hokamyuen/article/details/73330319?utm_source=blogkpcl2

考虑下签名？如果给别的openwrt配置这个仓库，会报签名失败。

root@OpenWrt:/tmp# opkg update
Downloading https://op.supes.top/packages/mipsel_24kc//Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages2
Downloading https://op.supes.top/packages/mipsel_24kc//Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://op.supes.top/snapshots/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://op.supes.top/snapshots/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_freifunk
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/freifunk/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.

garypang13, 2020-11-30:
这个本来就不是公共仓库,只是给本固件用的,里面的插件即使安装成功了,版本号对不上插件也是用不了的.

jveay, 2020-12-02:
大佬， x86的好像没有adguardhome的界面的。这边软件包安装了。但是服务那边没有。
[image]

garypang13, 2020-12-02:
更新软件列表

jveay, 2020-12-03:
更新了软件列表,有软件包了。但是提示adguardhome依赖没有。。。

hmsjy2017, 2020-12-08:
> 更新软件列表

插件仓库可以镜像吗？

hmsjy2017, 2020-12-08:
> > > 更新软件列表
> > 
> > 
> > 插件仓库可以镜像吗？
> 
> 可以,但是里面的插件会验证固件版本信息,都只适用于此仓库源码编译的固件,
支持rsync协议吗？

lalasou, 2020-12-26:
sqm qos 地址 https://github.com/tohojo/sqm-scripts
这个在荒野无灯固件有，挖矿必备感觉像没有挖矿
但是无灯不更新了。
晚上测试了很多k2p固件 就楼主和无灯的最好用 并且支持ipv6
希望添加这个插件谢谢
sqm

garypang13, 2020-12-26:
@lalasou 这个插件官方仓库就有,直接在后台软件包中搜sqm就能找到,

lalasou, 2020-12-26:
奇怪 我搜了 没有 ，我再刷下试试

Tiexcn, 2020-12-26:
kmod-tg3
是否能收录这个组件呢，有几块博通的网卡需要这个，之前的版本还能opkg install 安装，最新版本已经不能直接安装了。会提示版本不对。谢谢

garypang13, 2020-12-26:
已收录.

a545347837, 2021-01-01:
pptpd，官方仓库有，但是和内核不兼容；
frps可以安装但貌似打开会有问题

garypang13, 2021-01-01:
@a545347837 pptpd不建议使用了,frps没问题的,

a545347837, 2021-01-01:
> @a545347837 pptpd不建议使用了,frps没问题的,

另外我fork你的仓库，啥代码都没改使用workflow编译k2p时，在apply patch步骤中
patching file package/feeds/custom/luci-app-bypass/root/etc/init.d/bypass
Hunk #2 FAILED at 995.
报着错导致后面无法继续 请问你知道什么原因吗

a545347837, 2021-01-01:
> @a545347837 pptpd不建议使用了,frps没问题的,
我安装了frps，打开时报如下错：
RPCError
RPC call to uci/get failed with ubus code 4: 未找到资源
  at ClassConstructor.handleCallReply (https://10.0.0.1:1443
是不是需要再装其它东西？

garypang13, 2021-01-05:
@a545347837 frps已修复,卸载重装frps即可.

sanshu-rom, 2021-01-14:
希望加入luci-app-phtunnel内网花生壳使用 ，连接
https://github.com/zxlhhyccc/bf-package-master/tree/bf650c3e8daf1a6f92590911309047e214113246/zxlhhyccc/luci-app-phtunnel

garypang13, 2021-01-14:
> 希望加入luci-app-phtunnel内网花生壳使用 ，连接
> https://github.com/zxlhhyccc/bf-package-master/tree/bf650c3e8daf1a6f92590911309047e214113246/zxlhhyccc/luci-app-phtunnel
早就有了呀,

sanshu-rom, 2021-01-14:
但是好像没看到插件

garypang13, 2021-01-14:
> 但是好像没看到插件

没有R2S版,源码不支持.

sanshu-rom, 2021-01-14:
> > 但是好像没看到插件
> 
> 没有R2S版,源码不支持.

我在仓库里面看到了我是d2

sanshu-rom, 2021-01-14:
> > 但是好像没看到插件
> 
> 没有R2S版,源码不支持.

安装后提示无法启动日志空白

zkh1992, 2021-01-19:
UUgamebooster这个插件在r2s上无法安装，说所有仓库都找不到kernel依赖

Plume-yu, 2021-01-23:
> UUgamebooster这个插件在r2s上无法安装，说所有仓库都找不到kernel依赖

内核依赖不就一个tun吗，？这个有的啊

aenron, 2021-01-31:
能否加入qBittorrent？

garypang13, 2021-01-31:
> 能否加入qBittorrent？

一直就有.

aenron, 2021-01-31:
> > 能否加入qBittorrent？
> 
> 一直就有.

好吧，我只看到了第一页

hmsjy2017, 2021-02-02:
加入 neofetch
https://github.com/dylanaraps/neofetch
[image]

liuxiao0725, 2021-02-06:
r2s能不能解决下掉wan口的问题，每隔一段时间 wan口就会重新连接 有时候lan口变成百兆

doushang, 2021-02-07:
luci-app-services-wolplus
https://github.com/sundaqiang/openwrt-packages.git
luci-app-shortcutmenu
https://github.com/doushang/luci-app-shortcutmenu.git

hmsjy2017, 2021-02-08:
https://github.com/sjz123321/openwrt-neofetch

garypang13, 2021-02-08:
> https://github.com/sjz123321/openwrt-neofetch

试过了, 无法编译通过.

hmsjy2017, 2021-02-08:
> > https://github.com/sjz123321/openwrt-neofetch
> 
> 试过了, 无法编译通过.

可以用这个：https://github.com/garypang13/OpenWrt/issues/71#issuecomment-771625005

garypang13, 2021-02-08:
> > > https://github.com/sjz123321/openwrt-neofetch
> > 
> > 
> > 试过了, 无法编译通过.
> 
> 可以用这个：[#71 (comment)](https://github.com/garypang13/OpenWrt/issues/71#issuecomment-771625005)

这样没办法做成插件, 编译进系统的话就没必要了.

hmsjy2017, 2021-02-08:
> > > > https://github.com/sjz123321/openwrt-neofetch
> > > 
> > > 
> > > 试过了, 无法编译通过.
> > 
> > 
> > 可以用这个：[#71 (comment)](https://github.com/garypang13/OpenWrt/issues/71#issuecomment-771625005)
> 
> 这样没办法做成插件, 编译进固件的话就没必要了.

写一个 Makefile ？

[99 more replies]

closed 2021-12-29 by kiddin9