kiddin9/Kwrt #76: 又出了很多问题。

lx521603, 2020-06-17:
1.-ash: /etc/profile.d/sysinfo.sh: line 30: syntax error: unexpected "(" (expecting "then")  出现这个提示。

2. 这几天的保留配置升级，全不成功，升级后没任何配置。

3. 恢复备份后，adbyby+, smartdns,ssr等全部都有配置，不过运行不起来；但登录密码，IP地址，网关，DNS等配置是初始的，完全没有恢复成备份设置。

4.luci界面相当卡，每次过很多秒才能设置。

最终结果是，我这几天编译的都能进luci，但功能都运行不起来，不能使用。

kkkidding, 2020-06-17:
更新到最新代码, 问题1前两天就解决了
保留配置升级全不成功会不会是容量满了造成的.多分配点空间,软路由多的就是空间.
建议就用我编译好的固件,不包含插件,然后在后台在线安装自己想要的插件,一劳永逸.

lx521603, 2020-06-17:
> 更新到最新代码, 问题1前两天就解决了
> 保留配置升级全不成功会不会是容量满了造成的.多分配点空间,软路由多的就是空间.
> 建议就用我编译好的固件,不包含插件,然后在后台在线安装自己想要的插件,一劳永逸.

我昨晚pull到最新的编译的。

今天我看见又更新了不少，大概4小时前又pull最新的编译，还是一样的问题。

我没编译几个插件，空间确实给的不多，是20 160，平时都没问题。每次我看还剩下几十M剩余空间呢。

等我再试试。

lx521603, 2020-06-17:
> 更新到最新代码, 问题1前两天就解决了
> 保留配置升级全不成功会不会是容量满了造成的.多分配点空间,软路由多的就是空间.
> 建议就用我编译好的固件,不包含插件,然后在后台在线安装自己想要的插件,一劳永逸.

刚又pull了最新的，1的问题依然存在，还有有提示

2的问题依然存在，保留配置升级不行，配置会被清空。

3,4的问题没有了，我把空间从160增加到了280，同样的插件，同样的空间，以前绝对够的，不知道后来是增加了些什么。

garypang13, 2020-06-18:
问题1确实还有问题,已修复,临时解决方案:
执行: sed -i 's/\[ -e "$FILE" \] && . "$FILE"/[ -e "$FILE" ] \&\& env -i bash "$FILE"/g' /etc/profile
问题2 真没遇到过,每次编译出的固件我都有自己刷过测试的.

lx521603, 2020-06-20:
> 问题1确实还有问题,已修复,临时解决方案:
> 执行: sed -i 's/[ -e "$FILE" ] && . "$FILE"/[ -e "$FILE" ] && env -i bash "$FILE"/g' /etc/profile
> 问题2 真没遇到过,每次编译出的固件我都有自己刷过测试的.

感谢，我昨天pull最新的，编译后1,2都还继续出问题。
今早pull最新的编译后，所有问题都已经解决。

closed 2020-06-21 by garypang13