kiddin9/Kwrt #77: XXR plus 无法正常运行

tonyliangli, 2020-06-19:
提示：Can't find /usr/sbin/trojan program, can't start!
目录里也确实没有，麻烦查看一下啊，谢谢。

garypang13, 2020-06-20:
确实忘记加trojan到仓库了,晚点加上
掉网问题多观察几天,看是否会重复出现.

tonyliangli, 2020-06-20:
多谢，麻烦了。掉网我怀疑是否和 ipv6 有关？最新的固件虽然加入了 ipv6 的支持，但是/etc/sysctl.d/99-custom.conf 都是默认禁用的，我都打开之后，LAN 口才能获取 ipv6 地址，不然显示 undefined。

tonyliangli, 2020-06-20:
有的时候刚重启完 openwrt 就没有网络，需要到网络-接口-LAN-重新启动才可以。

garypang13, 2020-06-20:
有可能,ipv6我没环境所以并没有测试,

tonyliangli, 2020-06-20:
您方便时看看 trojan 可以通过软件仓库opkg安装吗？我记得貌似涉及 openssl，只能编译进固件，不是很清楚。

在2020年06月20日 10:32，garypang13 写道：

有可能,ipv6我没环境所以并没有测试,

lx521603, 2020-06-20:
> 提示：Can't find /usr/sbin/trojan program, can't start!
> 软件包里也确实没有，貌似只能通过编译安装？麻烦查看一下，谢谢。
> 另外发现会掉网，早上起来所有网页都打不开了，在网络-接口-重新启动LAN口（我是旁路）后恢复。

我10多天前编译的版本会出现类似的问题，但6月10号之后编译的就没问题了。

我的问题是早上起来基本什么网站都访问不了，解决办法是直接重启就交换机。我也不知道什么原理，搞得我都想交换机去售后了，但后面有没出现，我就怀疑是固件问题。

garypang13, 2020-06-20:
已添加trojan

closed 2020-06-21 by garypang13