kiddin9/Kwrt #81: IPV6无法获取地址和前缀

RUI98, 2020-06-22:
Mon Jun 22 09:14:33 2020 daemon.err pppd[19755]: sif6addr: ioctl(SIOCSIFADDR): Permission denied (line 2848)
Mon Jun 22 09:14:33 2020 daemon.warn pppd[19755]: cif6addr: ioctl(SIOCDIFADDR): No such address
Mon Jun 22 09:14:33 2020 daemon.notice netifd: Network device 'pppoe-WAN6' link is up
Mon Jun 22 09:14:33 2020 daemon.notice netifd: Interface 'WAN6' is now up
Mon Jun 22 09:14:33 2020 user.notice firewall: Reloading firewall due to ifup of WAN6 (pppoe-WAN6)
Mon Jun 22 09:14:33 2020 kern.warn kernel: [  267.947098] xt_FULLCONENAT: warning: failed to register a conntrack notifier. Disable active GC for mappings.
Mon Jun 22 09:14:48 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:15:04 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:15:20 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:15:36 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:15:52 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:16:08 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:16:24 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:16:40 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)
Mon Jun 22 09:16:56 2020 daemon.err odhcpd[3026]: Failed to send to ff02::1%lan@br-lan (Permission denied)

garypang13, 2020-06-22:
[image]
我的IPV6可以正确获取.

garypang13, 2020-06-22:
开启IPV6需要修改 /etc/sysctl.d/99-custom.conf 删掉
```
net.ipv6.conf.lo.disable_ipv6=1
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1 
```

closed 2020-06-24 by garypang13

whao123yahoo, 2024-04-29:
> 开启IPV6需要修改 /etc/sysctl.d/99-custom.conf 删掉
> 
> ```
> net.ipv6.conf.lo.disable_ipv6=1
> net.ipv6.conf.all.disable_ipv6=1
> net.ipv6.conf.default.disable_ipv6=1 
> ```
> 
> 后重启路由

我咋没有这个文件