kiddin9/Kwrt #82: K2P编译出来固件过大无法刷机

lexin8, 2020-06-23:
16M的闪存，编译出来接近16M无法完成刷机。
取消的插件包只留饮料都不够。看论坛都编译了很多插件，还能用主题，他们是怎么做精简的
[image]

garypang13, 2020-06-23:
把nginx换成uHTTPd试试. 另外5.4 内核在低端设备上可能会性能不足.

garypang13, 2020-06-23:
另外不要编译v2ray,v2ray是没有压缩的,解压后有17M,或者加上upx压缩.

lexin8, 2020-06-23:
[image]

lexin8, 2020-06-23:
有点奔溃，搞了一天还是各自问题。麻烦帮编译一个K2P 固件吧

需要igmpproxy、SSR、syncdial、无线驱动这四个，最好加一个网易云
固件15M以下，大于15.3就报错刷不进了。

garypang13, 2020-06-23:
我试试,

lexin8, 2020-06-23:
> 我试试,

这位做的固件插件挺多的，不知道怎么做到精简，给你借鉴一下，但是igmpproxy好像用的是LEDE的导致不能使用
https://github.com/lexin8/Phicomm-K2P-Openwrt-Autobuild

garypang13, 2020-06-24:
[06.24 phicomm_k2p.zip](https://github.com/garypang13/Actions-OpenWrt-Nginx/files/4822754/06.24.phicomm_k2p.zip)
刚好15M

lexin8, 2020-06-24:
> [06.24 phicomm_k2p.zip](https://github.com/garypang13/Actions-OpenWrt-Nginx/files/4822754/06.24.phicomm_k2p.zip)
> 刚好15M
> 你测试下,好用的话我考虑把k2p的自动编译也加进来.

[image]
initramfs-kernel.bin 内核低一起打包吧

garypang13, 2020-06-24:
命令行执行:  chmod +x /www/cgi-bin/luci

initramfs-kernel.bin [06.24 phicomm_k2p_2.zip](https://github.com/garypang13/Actions-OpenWrt-Nginx/files/4823356/06.24.phicomm_k2p_2.zip)

lexin8, 2020-06-24:
> 命令行执行: chmod +x /www/cgi-bin/luci
> 
> initramfs-kernel.bin [06.24 phicomm_k2p_2.zip](https://github.com/garypang13/Actions-OpenWrt-Nginx/files/4823356/06.24.phicomm_k2p_2.zip)

交换机功能没有了没办法划分VLAN，v2ray阉割了有点怪怪的

lexin8, 2020-06-24:
> > 命令行执行: chmod +x /www/cgi-bin/luci
> > initramfs-kernel.bin [06.24 phicomm_k2p_2.zip](https://github.com/garypang13/Actions-OpenWrt-Nginx/files/4823356/06.24.phicomm_k2p_2.zip)
> 
> 交换机功能没有了没办法划分VLAN，v2ray阉割了有点怪怪的

github.com/tjuyy/Phicomm-K2P-Openwrt-Autobuild这位编译的带有v2ray，交换机SSR还有其他插件，不知道怎么删减的，固件才14M左右，我编译单单带有无线驱动就15M了

garypang13, 2020-06-24:
交换机功能不是设置的问题吗,难道还有专门的模块
v2ray是百分百要阉割的,v2ray对配置要求很高的,就算硬加进去,速度跑起来也会很难看,去后台在线装个trojan吧

lexin8, 2020-06-24:
> 交换机功能不是设置的问题吗,难道还有专门的模块
> v2ray是百分百要阉割的,v2ray对配置要求很高的,就算硬加进去,速度跑起来也会很难看,去后台在线装个trojan吧

[image]
[image]

garypang13, 2020-06-24:
5.4 内核的不支持 VLAN 交换机配置

closed 2020-06-24 by garypang13

macaujack, 2024-05-13:
想请教一下，用 lede 代码 `make menuconfig` 时要怎么样选才能精简固件体积呀？方便的话可否分享下 `.config` 文件，十分感谢