kiddin9/Kwrt #85: docker和dockerman无法安装

clouded36, 2020-06-26:
显示kernel版本不兼容，强行安装的话，dockerman里只有一个概览，概览里没有任何信息，再软件包里看也显示docker和dockerman是未安装的状态。

garypang13, 2020-06-26:
[image]
我刚测试的,可以正常开启.
重装试试.

clouded36, 2020-06-26:
我用的6/26的最新的镜像全新安装的openwrt
首次通过软件包安装docker的时候，会一直卡在“正在执行软件包管理器”这个状态，并且没有任何信息，等了半个小时以上一直卡在这个状态，刷新页面后看dockerman里就是只有概览。
重新安装dockerman会出现

Collected errors:
 * opkg_conf_load: Could not lock /var/lock/opkg.lock: Resource temporarily unavailable.
通过命令

 rm -f /var/lock/opkg.lock

重置之后，重新安装dockerman同样还是卡“正在执行软件包管理器”，结果一样。
用opkg命令安装的话，就会出现

root@OpenWrt:~# opkg install docker-ce
Installing docker-ce (19.03.11-1) to root...
Downloading https://mirrors.cloud.tencent.com/openwrt/snapshots/packages/x86_64/packages/docker-ce_19.03.11-1_x86_64.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for docker-ce:
 * 	kernel (= 5.4.48-1-de1791898c4eaf32749663a1dc3fb362)
 * opkg_install_cmd: Cannot install package docker-ce.

显示无法安装。

不知道您是如何操作的，我的操作是哪里有问题吗？

garypang13, 2020-06-26:
命令行 opkg install --force-depends luci-app-dockerman 试试
我就是在后台安装的,只是等的时间稍微久点,

clouded36, 2020-06-26:
用命令行opkg install --force-depends luci-app-dockerman安装的话，会卡在Configuring docker-ce.强制取消后在luci里倒是能看到docker安装成功了，但是又出现个新的问题，就是无论安装什么容器，一但启动容器就会造成系统自动重启

garypang13, 2020-06-26:
我对docker不太熟,你试试编译的时候加上dockerman试试

garypang13, 2020-07-01:
最新版已修复docker无法启动容器的问题.

closed 2020-07-06 by garypang13