kiddin9/Kwrt #88: K2P安装了。但是。。。

Tint0ri, 2020-06-27:
1.没有2.4G
2.空间太小，SSR啥的都装不了。

是我配置有啥不对吗？直接用的0626 Release的版本。

garypang13, 2020-06-27:
开源驱动2.4G和5G不能同时开启
SSR plus才100多kb也装不下么,你登录ssh,  df -h 看看/overlay剩余空间是多少

Tint0ri, 2020-06-27:
能整合Lean的闭源驱动吗？至少2.4G和5G都有。
K2P的/overlay空间就是很小，不到1M。SSR，或者Passwall加上依赖包都超过了。

garypang13, 2020-06-27:
Lean的闭源驱动不支持最新的kernel 5.4
如果是ssr ssr协议的话不用再装其他依赖就可以使用,

Tint0ri, 2020-06-27:
> Lean的闭源驱动不支持最新的kernel 5.4
> 如果是ssr ssr协议的话不用再装其他依赖就可以使用,

可是用插件参考在线装SSR+的IPK的时候，系统报空间不够错误。

garypang13, 2020-06-29:
> > Lean的闭源驱动不支持最新的kernel 5.4
> > 如果是ssr ssr协议的话不用再装其他依赖就可以使用,
> 
> 可是用插件仓库在线装SSR+的IPK的时候，系统报空间不够错误。
> 
> 有办法调整overlay的空间大小吗？

在没有装任何其他插件的情况下也装不了ssr+吗
/overlay的总容量是多少,可用容量是多少

Tint0ri, 2020-06-30:
> > > Lean的闭源驱动不支持最新的kernel 5.4
> > > 如果是ssr ssr协议的话不用再装其他依赖就可以使用,
> > 
> > 
> > 可是用插件仓库在线装SSR+的IPK的时候，系统报空间不够错误。
> > 有办法调整overlay的空间大小吗？
> 
> 在没有装任何其他插件的情况下也装不了ssr+吗
> /overlay的总容量是多少,可用容量是多少

因为已经换Padavan固件了。所以具体数值记不太清了。大概是刷完有不到1M~1.5M。

garypang13, 2020-07-01:
固件缩小到了12.3 MB,应该能安装一些常用插件了,

Tint0ri, 2020-07-02:
> 固件缩小到了12.3 MB,应该能安装一些常用插件了,

谢谢。回头试试。
另外，现在闭源驱动支持5.4内核了。期待集成。
https://github.com/Lienol/openwrt/commit/5c1aa1dc65f11531c71c796ddab50eb8eed482fc

garypang13, 2020-07-02:
> > 固件缩小到了12.3 MB,应该能安装一些常用插件了,
> 
> 谢谢。回头试试。
> 另外，现在闭源驱动支持5.4内核了。期待集成。
> [Lienol/openwrt@5c1aa1d](https://github.com/Lienol/openwrt/commit/5c1aa1dc65f11531c71c796ddab50eb8eed482fc)

谢谢, 闭源驱动已集成.

closed 2020-07-03 by Tint0ri