kiddin9/Kwrt #98: 开启 ipv6 穿透模式 遇到问题

YoungWilliamZ, 2020-07-03:
参考：https://www.right.com.cn/FORUM/thread-2571254-1-1.html 配置ipv6 穿透模式 

卡在 安装 ebtables 和 kmod-ebtables-ipv6 模块

```
root@OpenWrt:~# opkg install ebtables
Installing ebtables (2018-06-27-48cff25d-1) to root...
Downloading https://mirrors.cloud.tencent.com/openwrt/snapshots/packages/x86_64/base/ebtables_2018-06-27-48cff25d-1_x86_64.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ebtables:
 * 	kernel (= 5.4.48-1-5721018b79f95e5064cb5dbe438cf5bd)
 * opkg_install_cmd: Cannot install package ebtables.
```

用 `opkg install ebtables --force-depends` 安装就会自动重启。

```
root@OpenWrt:~# opkg install ebtables --force-depends
Installing ebtables (2018-06-27-48cff25d-1) to root...
Downloading https://mirrors.cloud.tencent.com/openwrt/snapshots/packages/x86_64/base/ebtables_2018-06-27-48cff25d-1_x86_64.ipk
Installing kmod-ebtables (5.4.48-1) to root...
Downloading https://mirrors.cloud.tencent.com/openwrt/snapshots/targets/x86/64/packages/kmod-ebtables_5.4.48-1_x86_64.ipk
Configuring kmod-ebtables.
```

garypang13, 2020-07-04:
已解决,刷最新固件试试.

closed 2020-07-06 by garypang13

littlewhitelyw, 2021-01-12:
你好：
红米AC2100  2021.01.10版本
还是提示内核不兼容
`已安装的软件包 kernel 版本不兼容，要求 5.4.87-1-c47c04fe… 而 5.4.87-1-2d1ef64d… 已安装。`

garypang13, 2021-01-12:
@littlewhitelyw 下一版支持.