kiddin9/Kwrt #103: 反映下这几天的固件问题。

lx521603, 2020-07-05:
passwall（或者ssr+),smartdns这2个不会启动。但广告屏蔽大师这些正常运行。我没有全部检查。

重启也是一样的，这几个不会运行的还是不会运行。

唯一解决办法是去启动项里面，手动点击启动。（手动启动后，重启测试，重启后，这2个还是不会运行，还是需要去手动启动）

这个最大的问题是，万一断电重启后，这2个不启动，直接不能上网，望解决。

这3天，我每天都新pull了，每天都编译了新固件，都有这个问题。

前天的固件，我早上起来，这2个挂了，我只有重新手动启动。
昨天的固件，运行了几分钟后，这2个服务就挂了，又只有手动启动，然后到今天都没出现。
今天的固件，我才装上，还待验证。

garypang13, 2020-07-05:
启动项里 smartdns和ssr+是开机自启动状态吗
[image]
应该不会是固件的问题,并未有其他人反馈过类似的问题, 建议全新安装试试

lx521603, 2020-07-05:
> 启动项里 smartdns和ssr+是开机自启动状态吗
> [image]
> 应该不会是固件的问题,并未有其他人反馈过类似的问题, 建议全新安装试试

当时我进去看是全部都显示已经启动。我昨天就是全新安装的，没有恢复备份，手工配置了一遍。但几分钟就自动停了。只能手动启动，但重启后依旧没有运行。

所以我今天看你更新了不少，又才重新编译了最新固件，今天是升级安装(我只配置了passwall，因为其它你都配置好了，不用自己搞，用你的固件的最大原因就是这个，太省心了)，升级后smartdns，passwall挂掉，没有运行，重启没有解决，只能手动开启。

garypang13, 2020-07-06:
最好是直接用我编译好的固件,插件全部采用在线安装,目前群里并未有人发聩过固件有你说的这个问题.

lx521603, 2020-07-06:
> 最好是直接用我编译好的固件,插件全部采用在线安装,目前群里并未有人发聩过固件有你说的这个问题.

我直接用你编译好的固件，在线安装有点麻烦，v2ray这些都要单独安装，这个不说。

但我说的问题并没有完全解决，我只安装了一个passwall，配置好，总开关勾上，运用，它不运行，还是得跑到启动项里面手动开启。

并且还有个问题。后面我分别安装了adbyby，smartdns。那个汉化包是安装成功了的，所有的插件，内容是中文的，但所有菜单都是英文的。
比如passwall是这样：Basic SettingsNode ListAuto SwitchOther SettingsNode SubscribeRule UpdateAccess controlServer-SideWatch Logs，菜单下面的内容是正常中文。

之后，我重启了3次，插件是正常运行了。

结论是，直接用你的固件， 汉化不完全，passwall第一次不会运行.(也许它需要先重启一次？）

lx521603, 2020-07-06:
[image]

lx521603, 2020-07-06:
[image]

lx521603, 2020-07-06:
所有插件我都安装了最后那个汉化包，但汉化都不完全，如图。
但是插件直接编译进固件里就没问题。

garypang13, 2020-07-06:
在线安装只需要安装一次,以后每次系统升级都会自动安装
汉化不完整是浏览器缓存的问题
有插件第一次安装没自动运行的话用命令行或启动项里启动,或者重启下系统.是插件的问题,和固件没关系,也不影响以后的使用.

lx521603, 2020-07-06:
确实，试了下，是缓存问题。
我最后再编译测试下，如果自己编译的还是出现重启后不自动运行的话就用你编译的了，省得麻烦，谢谢了！

lx521603, 2020-07-06:
> 在线安装只需要安装一次,以后每次系统升级都会自动安装
> 汉化不完整是浏览器缓存的问题
> 有插件第一次安装没自动运行的话用命令行或启动项里启动,或者重启下系统.是插件的问题,和固件没关系,也不影响以后的使用.

3小时前，我最后编译了一次，无意中，我发现了问题所在。
我自己编译的，配置好以后，passwall,smartdns，配置好没启动，我就重启看看能不能解决，重启后，我看了下，还是没启动。

然后我没管，看了其它的网页去了，过了1,2分钟，我看smartdns竟然显示正在运行（我网页停留在这一页的），切换到passwall也是显示运行了。

我再重启系统，然后10多秒看，2个都没有启动，我就看状态栏，CPU100%，3个负载的数值都高，半分钟切换过去还是没运行。终端运行top显示passwall cpu占用率很高，等到CPU占用率降到正常地方，我切换过去看，2个都已经运行，但这个时候已经1分多了。我从来没遇到过。
我E3的CPU，设置的4核运行，比大部分人的软路由强很多。

至此，基本已经没什么问题了。

closed 2020-07-07 by garypang13