kiddin9/Kwrt #110: 我又来了

lx521603, 2020-07-16:
vsftpd openssh-sftp-server  这2个默认没有安装，之前的有。

Edge 主题下，改N次密码不生效，我就纳闷了，切换到bootstrap下解决。

garypang13, 2020-07-16:
vsftpd openssh-sftp-server 这两个都是有的,
Edge 主题改密码确实有点bug,需要改两次才会生效.后续会修复.

lx521603, 2020-07-16:
> vsftpd openssh-sftp-server 这两个都是有的,
> Edge 主题改密码确实有点bug,需要改两次才会生效.后续会修复.

我pull昨天的，确实没有，因为刷新固件后，我都要用wincp，传adbyby的广告过滤规则（adhome过滤不掉广告，不然我就adhome还省事），wincp提示没有sftp协议，我查了下缺这2个，然后我去软件包那里的“可用”里面安装后，正常使用。

smartdns运行一段时间后，自己挂了，重启也不能运行，又要手动跑到启动项里面去点启动。

garypang13, 2020-07-16:
默认确实没有安装,需要的话自己在软件包里安装下即可. 我都是用MobaXterm软件来上传,并不需要vsftpd openssh-sftp-server
我我的是SSR PLUS 从来没遇到过,smartdns挂掉的情况.

lx521603, 2020-07-18:
老大，问个问题，我在爱快上，设置DMZ，无论设到哪一个IP地址，都是生效的（我在爱快上绑定了阿里DDNS，输入域名就能知道成没成功），但你的固件不成功（设置虚拟机上的IP，爱快的IP都是成功的）。我同时做了端口映射，如果域名+端口号是成功的。

我真不知道哪里做错了。

我看了又看，跟其它固件的区别是，接口-硬件，里面，其它固件有 以太网适配器: "bond0"，难道是因为这个的原因？
我不知道它是干嘛的

garypang13, 2020-07-19:
这个我也不太清楚呢,

lx521603, 2020-07-20:
2020-07-20T05:49:42.5705865Z cc1: some warnings being treated as errors
2020-07-20T05:49:42.5776573Z make[5]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1/os_dep/linux/os_intfs.o] Error 1
2020-07-20T05:49:42.5777033Z scripts/Makefile.build:265: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1/os_dep/linux/os_intfs.o' failed
2020-07-20T05:49:42.5781658Z Makefile:1707: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1' failed
2020-07-20T05:49:42.5782478Z make[4]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1] Error 2
2020-07-20T05:49:42.5783050Z make[3]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1/.built] Error 2
2020-07-20T05:49:42.5787073Z make[2]: *** [package/feeds/custom/rtl8821cu/compile] Error 2
2020-07-20T05:49:42.5787475Z make[4]: Leaving directory '/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.52'
2020-07-20T05:49:42.5787837Z Makefile:74: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/rtl8821cu-5.8.1/.built' failed
2020-07-20T05:49:42.5788166Z make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/feeds/custom/luci/rtl8821cu'
2020-07-20T05:49:42.5788298Z time: package/feeds/custom/rtl8821cu/compile#1.63#0.41#2.05
2020-07-20T05:49:42.5790483Z package/Makefile:111: recipe for target 'package/feeds/custom/rtl8821cu/compile' failed
2020-07-20T05:49:42.5797236Z make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T05:49:42.5835627Z make[1]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2020-07-20T05:49:42.5836159Z package/Makefile:107: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-07-20T05:49:42.5837038Z make: *** [world] Error 2
2020-07-20T05:49:42.5843548Z make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T05:49:42.5845292Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
2020-07-20T05:49:42.5905044Z ##[error]Process completed with exit code 2.
2020-07-20T05:49:42.6077733Z Post job cleanup.
2020-07-20T05:49:43.6119260Z [command]/usr/bin/git version
2020-07-20T05:49:43.6182440Z git version 2.27.0
2020-07-20T05:49:43.6238376Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-07-20T05:49:43.6283019Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-07-20T05:49:43.6658642Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-07-20T05:49:43.6696997Z http.https://github.com/.extraheader
2020-07-20T05:49:43.6697614Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-07-20T05:49:43.6729252Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-07-20T05:49:43.7014838Z Cleaning up orphan processes

rtl8821cu驱动编译不过去？我怎么去除它？云编译我找不到在哪里

garypang13, 2020-07-20:
已修复

lx521603, 2020-07-20:
> 已修复

2020-07-20T15:52:08.5474726Z Configuring ppp-mod-pppoe.
2020-07-20T15:52:08.5474950Z Configuring coreutils-nohup.
2020-07-20T15:52:08.5475173Z Configuring luci-app-passwall.
2020-07-20T15:52:08.5475625Z  * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
2020-07-20T15:52:08.5475795Z 	But that file is already provided by package  * strongswan
2020-07-20T15:52:08.5476077Z  * opkg_install_cmd: Cannot install package luci-i18n-ipsec-vpnd-zh-cn.
2020-07-20T15:52:08.5476495Z  * check_data_file_clashes: Package luci-app-ipsec-vpnd wants to install file /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/ipsec
2020-07-20T15:52:08.5476653Z 	But that file is already provided by package  * strongswan
2020-07-20T15:52:08.5476927Z  * opkg_install_cmd: Cannot install package luci-app-ipsec-vpnd.
2020-07-20T15:52:08.5477238Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
2020-07-20T15:52:08.5477467Z  * 	chinadns-ng
2020-07-20T15:52:08.5477711Z  * opkg_install_cmd: Cannot install package luci-i18n-passwall-zh-cn.
2020-07-20T15:52:08.5478017Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
2020-07-20T15:52:08.5478249Z  * 	simple-obfs
2020-07-20T15:52:08.5478459Z  * 	v2ray-plugin
2020-07-20T15:52:08.5478909Z  * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
2020-07-20T15:52:08.5502368Z make[2]: *** [package/install] Error 255
2020-07-20T15:52:08.5502874Z package/Makefile:65: recipe for target 'package/install' failed
2020-07-20T15:52:08.5504201Z make[1]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
2020-07-20T15:52:08.5504807Z make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T15:52:08.5505645Z package/Makefile:108: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
2020-07-20T15:52:08.5512854Z make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T15:52:08.5531227Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
2020-07-20T15:52:08.5532780Z make: *** [world] Error 2
2020-07-20T15:52:08.5595099Z ##[error]Process completed with exit code 2.
2020-07-20T15:52:08.6164016Z Post job cleanup.
2020-07-20T15:52:09.0774241Z [command]/usr/bin/git version
2020-07-20T15:52:09.0839874Z git version 2.27.0
2020-07-20T15:52:09.0901807Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-07-20T15:52:09.0953191Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-07-20T15:52:09.1422729Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-07-20T15:52:09.1456834Z http.https://github.com/.extraheader
2020-07-20T15:52:09.1475990Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-07-20T15:52:09.1519992Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-07-20T15:52:09.1882124Z Cleaning up orphan processes

这次是ipsec VPN 出问题？前几天都还可以，我去除看看。

lx521603, 2020-07-21:
2020-07-20T19:13:02.0522295Z Configuring iptables-mod-fullconenat.
2020-07-20T19:13:02.0522563Z Configuring luci-i18n-firewall-zh-cn.
2020-07-20T19:13:02.0522809Z Configuring ppp-mod-pppoe.
2020-07-20T19:13:02.0523053Z Configuring coreutils-nohup.
2020-07-20T19:13:02.0523302Z Configuring luci-app-passwall.
2020-07-20T19:13:02.0523483Z make[2]: *** [package/install] Error 255
2020-07-20T19:13:02.0523870Z make[1]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
2020-07-20T19:13:02.0524192Z package/Makefile:65: recipe for target 'package/install' failed
2020-07-20T19:13:02.0524679Z make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T19:13:02.0525163Z package/Makefile:108: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
2020-07-20T19:13:02.0525525Z make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-20T19:13:02.0525907Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
2020-07-20T19:13:02.0526642Z make: *** [world] Error 2
2020-07-20T19:13:02.0578598Z ##[error]Process completed with exit code 2.
2020-07-20T19:13:02.0835860Z Post job cleanup.
2020-07-20T19:13:02.5506380Z [command]/usr/bin/git version
2020-07-20T19:13:02.5564841Z git version 2.27.0
2020-07-20T19:13:02.5623587Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-07-20T19:13:02.5680249Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-07-20T19:13:02.6122033Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-07-20T19:13:02.6157094Z http.https://github.com/.extraheader
2020-07-20T19:13:02.6166844Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-07-20T19:13:02.6199348Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-07-20T19:13:02.6518349Z Cleaning up orphan processes

然后

lx521603, 2020-07-21:
2020-07-21T14:05:58.0640218Z Configuring urngd.
2020-07-21T14:05:58.0640473Z Configuring iptables-mod-fullconenat.
2020-07-21T14:05:58.0681212Z Configuring luci-i18n-firewall-zh-cn.
2020-07-21T14:05:58.0681874Z Configuring ppp-mod-pppoe.
2020-07-21T14:05:58.0682481Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
2020-07-21T14:05:58.0682771Z  * 	simple-obfs
2020-07-21T14:05:58.0683948Z  * 	v2ray-plugin
2020-07-21T14:05:58.0684284Z  * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
2020-07-21T14:05:58.0684474Z make[2]: *** [package/install] Error 255
2020-07-21T14:05:58.0684925Z make[1]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
2020-07-21T14:05:58.0685281Z package/Makefile:65: recipe for target 'package/install' failed
2020-07-21T14:05:58.0685654Z make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-21T14:05:58.0686348Z package/Makefile:108: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
2020-07-21T14:05:58.0706109Z make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-21T14:05:58.0710098Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
2020-07-21T14:05:58.0712233Z make: *** [world] Error 2
2020-07-21T14:05:58.0792613Z ##[error]Process completed with exit code 2.
2020-07-21T14:05:58.0989684Z Post job cleanup.
2020-07-21T14:05:58.5635264Z [command]/usr/bin/git version
2020-07-21T14:05:58.5698699Z git version 2.27.0
2020-07-21T14:05:58.5752798Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2020-07-21T14:05:58.5809498Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2020-07-21T14:05:58.6277303Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2020-07-21T14:05:58.6321229Z http.https://github.com/.extraheader
2020-07-21T14:05:58.6331042Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2020-07-21T14:05:58.6375866Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2020-07-21T14:05:58.6697401Z Cleaning up orphan processes

我去，我的总会出问题

lx521603, 2020-07-21:
之前说的，爱快的DMZ指向openwrt的IP地址没有作用，我对比了下其它固件，
然后在 /etc/config/network 里加上了2段就没问题了
option _orig_ifname 'eth0'
option _orig_bridge 'true'

我的softether完全能使用了。
我不知道什么原理

garypang13, 2020-07-21:
@lx521603 softethervpn不能用不是那2段的问题,是kernel版本与软件包里的不配置导致的,新固件已经修复了,你今天肯定有更新了固件,

lx521603, 2020-07-21:
[image]
虽然我很想更新，但我一直没编译成功。现在正在使用的事17号，5天前编译成功的，运行了5天，没发现问题。
我现在是DMZ设到另外一个能运行softether vpn的固件上，那个固件只运行了这一个插件。
我就是对比那个固件的网络设置改了就正常了。

之前我不能验证是不是softethervpn能不能用，是dmz不起效啊，我用虚拟机里的ubuntu,ip，爱快的ip，unraid的ip，其它op的ip，都是秒生效。输入DDNS地址，马上就出现后台画面。 就是你的固件的不能访问。

这几天我就增加了这2段，它就没问题了。

lx521603, 2020-07-21:
但是，过了会，它又不行了。。。可能不是这2个代码的问题，重启了下，还是没用。

固件我去除了大部分插件，希望编译成功。

garypang13, 2020-07-21:
@lx521603

lx521603, 2020-07-21:
好的，了解

closed 2020-09-14 by garypang13