kiddin9/Kwrt #112: 关于AdGuardHome  默认配置的及过滤列表的一点反馈

Tiexcn, 2020-07-20:
多不久前开始用你的这个系统 ，基本上也都是用周未包，省事嘛，

用下来，AdGuardHome  总是我的一个必须关掉的功能，
即使是DNS 中过滤列表全取消/安全保护禁用掉，也总会有一些应用，一些网站，出现无法访问的情况 

默认状态下，虽然有屏蔽一些广告，但实际使用中确带来很多不便，比如其它第三方网站的 QQ快捷登录 打开缓慢，经常看论坛的话，这个功能就很重要了，电脑本机手动指运营商DNS秒开， 我LP公司的ERP网站，有几个就打不开，关掉AdGuardHome  秒开，还有一些其它正常的网站，也有机率发生这个问题，有一阵即使打开B站的速度 都有所影响。虽然我知道 到AD后台放行，可以解决这些问题，但这需要额外的操作，这离到手即用还有点差别，希望可以降低一点这个插件的管理范围。 其实 屏蔽广告这事，浏览器的插件做的更好，在非浏览器的地方，用这个东西效果可能有，但使用者感觉没有哪么明显。

另外 最新版本里的主题，我个人感觉 没有之前版本顺畅， SSR PLUS插件 订阅栏，好像也有兼容性问题，而且 不兼容WIN10 IE11浏览器，登录不了，密码都输入不了。

其实有个smartDNS 我感觉 就够用了。SSR PLUS 的DNS解析 默认指定的又是AdGuardHome且不能修改。或许可以给一个指定 SMARTDNS的选项，也是一个不错的方法，。

这是我目前使用过程中 碰到的一些问题，我的碎碎念先到这吧，有啥其它的我再反馈。感谢你所做出的努力。3Q

garypang13, 2020-07-20:
AdGuardHome我一直在用,能过滤掉50%以上的广告和无用DNS解析,主要我很少上国内的网站,所以感觉不出你说的这些问题, 你可以把 Energized Protection Blu这个规则关闭换成dGuard Simplified Domain Names试试, Energized Protection Blu误杀国内网站有点严重.
新主题登录页面的动画流畅度取决于浏览器以及电脑配置,其他页面不存在流畅度的差别.
IE11我确实没测试,我的WIN10默认浏览器是Edge,没有IE测试环境, 软微也彻底放弃了IE了
SSR PLUS插件 订阅栏这个问题已经解决,今天编译的版本会修复这个问题.

SSR PLUS 的DNS解析 默认指定的是5335端口,你不装adguardhome,把smartdns端口改为5335就行了.

Tiexcn, 2020-07-21:
更改了smartdns的商品号后，问题解决。3Q

closed 2020-07-31 by garypang13