kiddin9/Kwrt #115: x86_64 编译失败

bingxueqingzhi, 2020-07-29:
以下是失败的LOG, 谢谢

2020-07-29T02:14:07.0401388Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl-nginx:
2020-07-29T02:14:07.0401998Z  * 	nginx-util (>= 1.4-2)
2020-07-29T02:14:07.0402613Z  * opkg_install_cmd: Cannot install package luci-ssl-nginx.
2020-07-29T02:14:07.0403084Z make[2]: *** [package/install] Error 255
2020-07-29T02:14:07.0404579Z package/Makefile:65: recipe for target 'package/install' failed
2020-07-29T02:14:07.0410116Z make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-29T02:14:07.0454269Z make[1]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
2020-07-29T02:14:07.0454905Z package/Makefile:108: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
2020-07-29T02:14:07.0457009Z make: *** [world] Error 2
2020-07-29T02:14:07.0457481Z make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt'
2020-07-29T02:14:07.0462462Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
2020-07-29T02:14:07.0579815Z ##[error]Process completed with exit code 2.

bingxueqingzhi, 2020-07-31:
https://pipelines.actions.githubusercontent.com/JGwNj1SkbZB5NVb9le53PfDY7V7StTcMXjEtmWu4q291asgbqK/_apis/pipelines/1/runs/9/signedlogcontent/4?urlExpires=2020-07-31T01%3A19%3A25.9466638Z&urlSigningMethod=HMACV1&urlSignature=9j9hT%2FqZ28pi757lxdxDH9%2FPLsXhywOTG9eY284TqCs%3D

能帮忙看看x86-64是怎么回事吗

garypang13, 2020-07-31:
已经修复了,同步最新代码

closed 2020-07-31 by garypang13

bingxueqingzhi, 2020-10-28:
x86-64和R2S编译失败, 能帮忙看看是什么原因吗

bingxueqingzhi, 2020-10-28:
> 
> 
> 已经修复了,同步最新代码

x86-64和R2S编译失败, 能帮忙看看是什么原因吗

garypang13, 2020-10-28:
> > 已经修复了,同步最新代码
> 
> x86-64和R2S编译失败, 能帮忙看看是什么原因吗

已修复,pull最新代码.