kiddin9/Kwrt #116: k2p 的 openssl 有进行O3优化吗？

moxuanyuan, 2020-07-30:
我看 chongshengB 的padavan 和 Lean openwrt 源码 commits 中都有提到对openssl 进行优化，不知这个项目有没有同样优化？

kkkidding, 2020-07-30:
我对比了下,并没有特别需要优化的.你可以指明具体哪条commit,我再具体看下.

garypang13, 2020-07-30:
@moxuanyuan 我对比了下,并没有特别需要优化的.你可以指明具体哪条commit,我再具体看下.

moxuanyuan, 2020-07-31:
chongshengB padavan commit  https://github.com/chongshengB/rt-n56u/commit/401751c91347e4ca7c633777fe2096c7bf516db2
Lean openwrt commit    https://github.com/coolsnowwolf/lede/commit/3b9a77b9b02d8fbc4ed2ca895556a93a491f8599#diff-b1cb346c37a66309731d582d2069075a
我是菜鸟，我简单对比了一下 官方 openwrt 源码 和 lean 源码，差别在于package/libs/openssl/Config.in 中的“OPENSSL_OPTIMIZE_SPEED”，官方是x86的默认开启，lean是直接默认开启。
对于k2p，你的源码是采用官方的，驱动则采用lean大的。所以我理解就是k2p的openssl没用进行O3优化。

garypang13, 2020-07-31:
> chongshengB padavan commit [chongshengB/rt-n56u@401751c](https://github.com/chongshengB/rt-n56u/commit/401751c91347e4ca7c633777fe2096c7bf516db2)
> Lean openwrt commit [coolsnowwolf/lede@3b9a77b#diff-b1cb346c37a66309731d582d2069075a](https://github.com/coolsnowwolf/lede/commit/3b9a77b9b02d8fbc4ed2ca895556a93a491f8599#diff-b1cb346c37a66309731d582d2069075a)
> 我是菜鸟，我简单对比了一下package/libs/openssl/Config.in ， 官方 [openwrt 源码](https://github.com/openwrt/openwrt/blob/master/package/libs/openssl/Config.in) 和 [lean 源码](https://github.com/coolsnowwolf/lede/blob/master/package/libs/openssl/Config.in)，对是openssl 优化差别“OPENSSL_OPTIMIZE_SPEED”，官方是x86的默认开启，lean是直接默认开启。
> 对于k2p，你的源码是采用官方的，驱动则采用lean大的。所以我理解就是k2p的openssl没用进行O3优化。

这个可以在.config文件中添加 CONFIG_OPENSSL_OPTIMIZE_SPEED=y 即可.

moxuanyuan, 2020-07-31:
> 这个可以在.config文件中添加 CONFIG_OPENSSL_OPTIMIZE_SPEED=y 即可.
明白了，谢谢！

closed 2020-07-31 by garypang13