kiddin9/Kwrt #118: x86_64尝试直接把帕斯窝编译进去, 失败

bingxueqingzhi, 2020-08-03:
以下是LOG里报错的部分:

2020-07-31T09:59:00.2198968Z Collected errors:
2020-07-31T09:59:00.2199674Z Configuring block-mount.
2020-07-31T09:59:00.2201217Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
2020-07-31T09:59:00.2201530Z Configuring kmod-usb-storage-extras.
2020-07-31T09:59:00.2202533Z  * 	chinadns-ng
2020-07-31T09:59:00.2202801Z Configuring kmod-nls-cp437.
2020-07-31T09:59:00.2203845Z  * opkg_install_cmd: Cannot install package luci-i18n-passwall-zh-cn.
2020-07-31T09:59:00.2204164Z Configuring kmod-nls-iso8859-1.
2020-07-31T09:59:00.2205098Z make[2]: *** [package/install] Error 255

liuran001, 2020-08-03:
chinadns-ng没有编译进去

> 2020-07-31T09:59:00.2201217Z * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn: 
2020-07-31T09:59:00.2202533Z * chinadns-ng

bingxueqingzhi, 2020-08-03:
> 
> 
> chinadns-ng没有编译进去
> 
> > 2020-07-31T09:59:00.2201217Z * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
> > 2020-07-31T09:59:00.2202533Z * chinadns-ng

求教如何解决

garypang13, 2020-08-03:
你是在本地编译的么,我看你github并没有这个报错

liuran001, 2020-08-03:
> > chinadns-ng没有编译进去
> > > 2020-07-31T09:59:00.2201217Z * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-passwall-zh-cn:
> > > 2020-07-31T09:59:00.2202533Z * chinadns-ng
> 
> 求教如何解决

config加上CONFIG_PACKAGE_chinadns-ng=y

bingxueqingzhi, 2020-08-03:
> 
> 
> 你是在本地编译的么,我看你github并没有这个报错

这是7/31的LOG,今天的还在编译中....

bingxueqingzhi, 2020-08-03:
> 
> 
> 你是在本地编译的么,我看你github并没有这个报错

云编译的, 这是7/31的LOG, 今天的还在编译中...

garypang13, 2020-08-03:
> > 你是在本地编译的么,我看你github并没有这个报错
> 
> 云编译的, 这是7/31的LOG, 今天的还在编译中...

已修复

bingxueqingzhi, 2020-08-04:
> 
> 
> > > 你是在本地编译的么,我看你github并没有这个报错
> > 
> > 
> > 云编译的, 这是7/31的LOG, 今天的还在编译中...
> 
> 已修复

已成功, 谢谢

closed 2020-08-05 by bingxueqingzhi