kiddin9/Kwrt #123: GitHub Action编译RM2100时Error 2

Magic-Xin, 2020-08-11:
commit是最新的f8ecf42a8be6b3ec1bf1ace51a228c580626dd87

> make[5]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/ap/ap_cfg.o] Error 1
scripts/Makefile.build:265: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/ap/ap_cfg.o' failed
make[4]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap] Error 2
Makefile:1707: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap' failed
make[4]: Leaving directory '/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.52'
Makefile:228: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/.built' failed
make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/feeds/custom/luci/mt-drivers/mt_wifi'
make[3]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/.built] Error 2
time: package/feeds/custom/mt_wifi/compile#1.52#0.45#1.91
package/Makefile:111: recipe for target 'package/feeds/custom/mt_wifi/compile' failed
make[2]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
make[2]: *** [package/feeds/custom/mt_wifi/compile] Error 2
package/Makefile:107: recipe for target '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt'
/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/include/toplevel.mk:233: recipe for target 'world' failed
make[1]: *** [/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make: *** [world] Error 2
##[error]Process completed with exit code 2.

garypang13, 2020-08-11:
已修复

Magic-Xin, 2020-08-11:
效率好高，感谢

Magic-Xin, 2020-08-11:
在 commet: 1a3320d5ca7cb1ae22e258c8998203e3d35cc718 改动后
redmi_ac2100 成功编译
但是刷机后发现没有无线，也没有安装 luci-app-mtwifi
手动安装后有无线页面，但是无法设置，缺失相关驱动

garypang13, 2020-08-12:
不好意思,已修复.

closed 2020-09-03 by Magic-Xin