kiddin9/Kwrt #129: 自行安装的插件有问题

LoogCP, 2020-08-27:
之前一直用的lean大佬的项目编译的固件，这次使用你的固件，软件里面自行安装ssr-plus出现无法启动的情况，
ssh进去卸载软件中心的ssr-plus相关，重新安装我编译的ssr-plus也不能启动

另外，主题有单独的吗，我觉得这个主题好漂亮，但是我把edge主题编译进去之后有问题，所有页面一片白，啥东西都没有

garypang13, 2020-08-27:
如果使用v2ray或trojan协议的话,还需要额外安装这两项
主题 https://github.com/garypang13/luci-theme-edge/ 注意如果是lean源码的话请选择18.06分支

chinapee, 2020-08-27:
用我编译的，没问题，这是一个分支，https://github.com/chinapee/Actions-OpenWrt/releases

LoogCP, 2020-08-27:
> 如果使用v2ray或trojan协议的话,还需要额外安装这两项
> 主题 https://github.com/garypang13/luci-theme-edge/ 注意如果是lean源码的话请选择18.06分支
感谢大佬！
我最开始是使用的trojan节点，然后不能启动就换ssr节点试了一下，发现还是不行。用自己编译的ipk就试了一下trojan，发现不能启动就刷回lean大佬的固件了
我再去试试18.06分支
等以后有时间了再折腾

LoogCP, 2020-08-27:
> 用我编译的，没问题，这是一个分支，https://github.com/chinapee/Actions-OpenWrt/releases

感谢大佬，
这段时间还是先用着lean大佬的固件，等以后有时间了再来折腾这个

LoogCP, 2020-08-27:
> 如果使用v2ray或trojan协议的话,还需要额外安装这两项
> 主题 https://github.com/garypang13/luci-theme-edge/ 注意如果是lean源码的话请选择18.06分支

我发现你的固件好像性能要比lean大佬的固件好些，我用lean大佬的固件的时候，使用网页那个升级，上传固件大概只有6、70M/s,但是你的固件能跑到120M/s,只不过上传到100%会卡住不动，最后我是在ssh里面升级的

LoogCP, 2020-08-28:
> 用我编译的，没问题，这是一个分支，https://github.com/chinapee/Actions-OpenWrt/releases

大佬，你这固件也太大了吧，5个G,没法直接刷进去，又不想用U盘

cqboys, 2020-09-02:
> > 如果使用v2ray或trojan协议的话,还需要额外安装这两项
> > 主题 https://github.com/garypang13/luci-theme-edge/ 注意如果是lean源码的话请选择18.06分支
> 
> 我发现你的固件好像性能要比lean大佬的固件好些，我用lean大佬的固件的时候，使用网页那个升级，上传固件大概只有6、70M/s,但是你的固件能跑到120M/s,只不过上传到100%会卡住不动，最后我是在ssh里面升级的

我也是同样的问题  只能在SSH 里面升级

closed 2020-10-03 by garypang13