kiddin9/Kwrt #148: 无法转发ftp至内网

WROIATE, 2020-09-27:
使用socat转发出现以下情况
[image]
请教一下如何转发ipv6的ftp至内网nas上

garypang13, 2020-09-27:
ipv4的ftp能成功吗,我用的IPV4没有问题.

WROIATE, 2020-09-27:
> ipv4的ftp能成功吗,我用的IPV4没有问题.

ipv4我能转发21端口的，如果我把端口换成别的也不行，同样会出现上面的积极拒绝
比如这种
[image]

随带说一下，他的端口的确是转进去了，我这能显示输入登录的信息，但是输入完就会出现上面的这种reset

WROIATE, 2020-09-27:
我记得原来用lean的好像没这问题，在lienol那也发现过这个情况

garypang13, 2020-09-27:
我这21或其他端口都可以,如果是其他端口比如1666的话,连接的时候客户端那边的FTP端口记得也要填1666.

WROIATE, 2020-09-27:
> 我这21或其他端口都可以,如果是其他端口比如1666的话,连接的时候客户端那边的FTP端口记得也要填1666.

我这改了端口号的，也确实不行
[image]

closed 2020-10-03 by garypang13