kiddin9/Kwrt #152: 能否添加斐讯n1的自动编译

WROIATE, 2020-10-06:
感觉现在玩n1比较多，可以考虑下吗

garypang13, 2020-10-07:
已添加N1,麻烦帮忙测试下.
https://op.supes.top/firmware/N1/

WROIATE, 2020-10-08:
> 已添加N1,麻烦帮忙测试下.
> https://op.supes.top/firmware/phicomm-N1/

直接访问luci显示Nginx 500
ssh使用用户root和密码root无法登陆，显示没有权限

garypang13, 2020-10-08:
有没可能是刷机方法不对导致的问题.
有测试过别的openwrt固件能正常使用吗.

WROIATE, 2020-10-08:
> 有没可能是打包生成刷机固件的时候出错了导致的.
> 具体打包可以参考下这个 https://www.youtube.com/watch?v=K-kreq93LqM

应该不是，我使用lean的源码编译打包是正常的

garypang13, 2020-10-09:
用lean源码重新编译了
https://op.supes.top/firmware/phicomm-N1/openwrt-armvirt-64-default-rootfs.tar.gz

WROIATE, 2020-10-09:
> 用lean源码重新编译了
> https://op.supes.top/firmware/phicomm-N1/openwrt-armvirt-64-default-rootfs.tar.gz

测试发现很容易崩溃，就是能够进入设置界面，一两分钟后就炸了，变成nginx 502，也无法ping通
并且不带ipv6，turboacc加速也有两个不同的在luci界面里

garypang13, 2020-10-09:
感谢反馈, ipv6，turboacc这两个问题很好解决
没在根目录找到安装进emmc的脚本,这个应该是打包工具提供的,在打包的时候就会打包进去才对.
至于很容易崩溃,只能把已知的问题都修复下再看看.

garypang13, 2020-10-09:
https://op.supes.top/firmware/phicomm-N1/10.09-openwrt-armvirt-64-default-rootfs.tar.gz
这一版如果还有问题那我也没办法了.不过IPV6这一版又被我给漏了...

closed 2020-10-16 by garypang13