kiddin9/Kwrt #165: 一直提示：cp: cannot create regular file 'openwrt/package/default-settings/files/zzz-default-settings': No such file or directory

ledewrt, 2020-10-17:
大佬：编译停留在这里报错
autobild的源码是这里： 
 if [ -f "devices/common/default-settings" ]; then
         cp -f devices/common/default-settings package/*/*/default-settings/files/zzz-default-settings
        fi
        if [ -f "devices/${{matrix.target}}/default-settings" ]; then
         cat devices/${{matrix.target}}/default-settings >> package/*/*/default-settings/files/zzz-default-settings

报错提示：cp: cannot create regular file 'openwrt/package/default-settings/files/zzz-default-settings': No such file or directory

garypang13, 2020-10-17:
openwrt/package/default-settings/files/zzz-default-settings
把package前的openwrt去掉,改为package/default-settings/files/zzz-default-settings

ledewrt, 2020-10-17:
去掉也不行。你的原版是没有的，也提示不存在。后来才加上也不存在。所以来请教下。

garypang13, 2020-10-17:
去掉一定可以.

closed 2020-10-20 by ledewrt