kiddin9/Kwrt #168: 一键脚本编译失败

WROIATE, 2020-10-18:
```
ccache: error: Failure running compiler check command: %compiler% -dumpmachine; %compiler% -dumpversion
[9/2073] ACTION //base:build_date(//build/toolchain/linux:clang_x64_openwrt)
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:89: /home/jarao/openwrt/build_dir/target-x86_64_musl/naiveproxy-86.0.4240.75-1/.built] Error 1
make[3]: Leaving directory '/home/jarao/openwrt/feeds/custom/naiveproxy'
time: package/feeds/custom/naiveproxy/compile#20.10#4.90#45.87
make[2]: *** [package/Makefile:114: package/feeds/custom/naiveproxy/compile] Error 2
make[2]: Leaving directory '/home/jarao/openwrt'
make[1]: *** [package/Makefile:108: /home/jarao/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jarao/openwrt'
make: *** [/home/jarao/openwrt/include/toplevel.mk:236: world] Error 2
```

garypang13, 2020-10-19:
编译naiveproxy的时候报错了,建议先取消naiveproxy

WROIATE, 2020-10-19:
v2ray编译也有问题
```
go: google.golang.org/grpc@v1.33.0: reading https://goproxy.io/google.golang.org/grpc/@v/v1.33.0.mod: 404 Not Found
        server response: not found: google.golang.org/grpc@v1.33.0: invalid version: unknown revision v1.33.0

Building targets
go: google.golang.org/grpc@v1.33.0: reading https://goproxy.io/google.golang.org/grpc/@v/v1.33.0.mod: 404 Not Found
        server response: not found: unknown revision v1.33.0

make[3]: *** [Makefile:312: /home/jarao/openwrt/build_dir/target-x86_64_musl/v2ray-core-4.31.1/.built] Error 1
make[3]: Leaving directory '/home/jarao/openwrt/feeds/custom/v2ray'
time: package/feeds/custom/v2ray/compile#0.31#0.17#9.32
make[2]: *** [package/Makefile:114: package/feeds/custom/v2ray/compile] Error 2
make[2]: Leaving directory '/home/jarao/openwrt'
make[1]: *** [package/Makefile:108: /home/jarao/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jarao/openwrt'
make: *** [/home/jarao/openwrt/include/toplevel.mk:236: world] Error 2
```

garypang13, 2020-10-20:
我最近编译的几个固件都默认包含v2ray的,都没有报错.

closed 2020-10-20 by WROIATE