kiddin9/Kwrt #172: 无法编译安装luci-app-socat

WROIATE, 2020-10-21:
选择了luci里的socat，莫名无法安装，请问源码是否有重复的luci-app-socat包？
```
Collected errors:
2020-10-21T06:36:32.1171293Z  * check_data_file_clashes: Package luci-app-socat wants to install file /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat
2020-10-21T06:36:32.1172332Z 	But that file is already provided by package  * socat
2020-10-21T06:36:32.1173663Z  * check_data_file_clashes: Package luci-app-socat wants to install file /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/socat
2020-10-21T06:36:32.1174685Z 	But that file is already provided by package  * socat
2020-10-21T06:36:32.1175482Z  * opkg_install_cmd: Cannot install package luci-i18n-socat-zh-cn.
2020-10-21T06:36:32.1176894Z  * check_data_file_clashes: Package luci-app-socat wants to install file /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/config/socat
2020-10-21T06:36:32.1177930Z 	But that file is already provided by package  * socat
2020-10-21T06:36:32.1179230Z  * check_data_file_clashes: Package luci-app-socat wants to install file /home/runner/work/Actions-OpenWrt/Actions-OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/socat
2020-10-21T06:36:32.1180248Z 	But that file is already provided by package  * socat
2020-10-21T06:36:32.1180982Z  * opkg_install_cmd: Cannot install package luci-app-socat.
```

garypang13, 2020-10-21:
luci-app-socat与官方packages中的socat不兼容,选上socat就行,没必要装luci-app-socat

WROIATE, 2020-10-21:
> luci-app-socat与官方packages中的socat不兼容,选上socat就行,没必要装luci-app-socat

我勾luci-app-socat是因为他有luci设置界面，如果不勾选也有吗？

WROIATE, 2020-10-26:
试了一下不勾选的话没有luci界面

garypang13, 2020-10-28:
已修复,可以勾选luci-app-socat了

WROIATE, 2020-10-28:
> 已修复,可以勾选luci-app-socat了

了解

closed 2020-10-28 by WROIATE