kiddin9/Kwrt #175: 在线安装提示ipk长度不匹配问题和内核版本不一致问题

ffii4455, 2020-10-29:
只能通过ssh强行安装吗？

garypang13, 2020-10-29:
更新列表, 内核版本不一致说明固件暂不支持此插件

ffii4455, 2020-10-29:
vpnd也不支持吗？强制安装可以运行起来

garypang13, 2020-10-29:
@ffii4455 vpnd并不会提示内核版本不一致

ffii4455, 2020-10-29:
@garypang13 
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ipsec-vpnd:
 * 	kernel (= 5.4.72-1-581e3be1bf0df812542bd1683dc1ef03)
 * opkg_install_cmd: Cannot install package luci-app-ipsec-vpnd.

我用的是周更版本内核是5.4.72，应该自己编译一个最新的？

garypang13, 2020-10-29:
是什么设备,我在x86设备上测试是可以正常安装luci-app-ipsec-vpnd

ffii4455, 2020-10-29:
路由器，小米AC2100

garypang13, 2020-10-29:
@ffii4455 ac2100下vpnd安装确实有问题,下一版修复.

ffii4455, 2020-10-29:
谢谢大佬 @garypang13

ffii4455, 2020-10-29:
@garypang13 大佬，实测opkg update后 长度报错依然存在。还请一起关注一下，谢谢！
Collected errors:
 * opkg_install_pkg: Package size mismatch: luci-app-openvpn-server is 21252 bytes, expecting 21257 bytes
 * opkg_install_cmd: Cannot install package luci-app-openvpn-server.

garypang13, 2020-10-30:
新版已添加vpnd支持.

IveanEx, 2020-11-07:
@garypang13 AC2100安装openclash也有一些问题，提示内核的版本不匹配。强制安装同样也没有任何问题

garypang13, 2020-11-07:
> @garypang13 AC2100安装openclash也有一些问题，提示内核的版本不匹配。强制安装同样也没有任何问题

下一版修复

closed 2021-12-29 by kiddin9