kiddin9/Kwrt #180: 大佬可以贴一下你的nginx.conf关于https这块是如何配置的吗

ryanxia0226, 2020-11-08:
我尝试在网上相关问题，试了一下但是都不成功，感谢了。

garypang13, 2020-11-08:
```
server {
  listen 80;
  listen 443 ssl http2;
  #listen [::]:443 ssl;
  server_name supes.top www.supes.top;
  index nginx.html index.html index.htm index.php default.html default.htm default.php;
  root  /www;
  access_log off;
		
  ssl_certificate /etc/acme/supes.top/fullchain.cer;
  ssl_certificate_key /etc/acme/supes.top/supes.top.key;
  ssl_protocols TLSv1 TLSv1.1 TLSv1.2 TLSv1.3;
  ssl_ciphers TLS13-AES-256-GCM-SHA384:TLS13-CHACHA20-POLY1305-SHA256:TLS13-AES-128-GCM-SHA256:TLS13-AES-128-CCM-8-SHA256:TLS13-AES-128-CCM-SHA256:EECDH+CHACHA20:EECDH+AES128:RSA+AES128:EECDH+AES256:RSA+AES256:EECDH+3DES:RSA+3DES:!MD5;
  ssl_prefer_server_ciphers on;
  ssl_session_timeout 10m;
  ssl_session_cache builtin:1000 shared:SSL:10m;
  ssl_buffer_size 1400;
  add_header Strict-Transport-Security max-age=15768000;
  ssl_stapling on;
  ssl_stapling_verify on;

  if ($ssl_protocol = "") { return 301 https://$host$request_uri; }
  if ($host = 'www.supes.top') {
	return 301 https://supes.top$request_uri;
  }
  #include rewrite/other.conf;
  #error_page 404 /404.html;
  #error_page 502 /502.html;

  location ~ .*\.(wma|wmv|asf|mp3|mmf|zip|rar|jpg|gif|png|swf|flv|mp4)$ {
    valid_referers none blocked supes.top *.supes.top op.supes.top;
    if ($invalid_referer) {
        return 403;
    }
  }

  location ~ .*\.(gif|jpg|jpeg|png|bmp|swf|flv|mp4|ico)$ {
    expires 30d;
    access_log off;
  }
  location ~ .*\.(js|css)?$ {
    expires 7d;
    access_log off;
  }
  location ~ /(\.user\.ini|\.ht|\.git|\.svn|\.project|LICENSE|README\.md) {
    deny all;
  }
		
	include conf.d/*.locations;

}
```

ryanxia0226, 2020-11-08:
谢谢👌

closed 2020-11-08 by ryanxia0226