kiddin9/Kwrt #187: 无法安装python

cnjn, 2020-11-15:
红米ac2100在使用opkg安装python3时报错，可能是软件源有问题。。。
```bash
root@OpenWrt:~# opkg update
Downloading https://op.supes.top/packages/mipsel/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_custom
Downloading https://op.supes.top/19.07.4/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages2
Downloading https://op.supes.top/snapshots/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_freifunk
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://op.supes.top/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
root@OpenWrt:~# opkg install python3
Package libavahi-dbus-support version 0.8-1 has no valid architecture, ignoring.
Package libavahi-nodbus-support version 0.8-1 has no valid architecture, ignoring.
Package libcap version 2.27-1 has no valid architecture, ignoring.
Package libdaemon version 0.14-5 has no valid architecture, ignoring.
Package libdbus version 1.12.12-1 has no valid architecture, ignoring.
Package libexpat version 2.2.9-1 has no valid architecture, ignoring.
Package libgd version 2.2.5-2 has no valid architecture, ignoring.
Package liblz4-1 version 1.9.2-4 has no valid architecture, ignoring.
Package libmpdclient version 2.18-1 has no valid architecture, ignoring.
Package libopus version 1.3.1-2 has no valid architecture, ignoring.
Package libpcre2-16 version 10.32-1 has no valid architecture, ignoring.
Package libpcre version 8.43-1 has no valid architecture, ignoring.
Package libsqlite3-0 version 3310100-1 has no valid architecture, ignoring.
Package libxml2 version 2.9.9-3 has no valid architecture, ignoring.
Package mariadb-common version 10.2.33-1 has no valid architecture, ignoring.
Package pulseaudio-daemon version 12.2-3 has no valid architecture, ignoring.
Package python3 version 3.7.9-1 has no valid architecture, ignoring.
Package python version 2.7.18-3 has no valid architecture, ignoring.
Package sbc version 1.4-2 has no valid architecture, ignoring.
Package shine version 3.1.1-1 has no valid architecture, ignoring.
Package strongswan version 5.8.2-2 has no valid architecture, ignoring.
Package yajl version 2.1.0-2 has no valid architecture, ignoring.
Package alsa-lib version 1.1.8-1 has no valid architecture, ignoring.
Package avahi-dbus-daemon version 0.8-1 has no valid architecture, ignoring.
Package boost-system version 1.71.0-6 has no valid architecture, ignoring.
Package boost version 1.71.0-6 has no valid architecture, ignoring.
Package coreutils-install version 8.30-2 has no valid architecture, ignoring.
Package coreutils version 8.30-2 has no valid architecture, ignoring.
Package dbus version 1.12.12-1 has no valid architecture, ignoring.
Package gcc version 7.4.0-5 has no valid architecture, ignoring.
Package glib2 version 2.58.3-5 has no valid architecture, ignoring.
Package libacl version 2.2.53-1 has no valid architecture, ignoring.
Package libattr version 2.4.48-1 has no valid architecture, ignoring.
Package libavahi-client version 0.8-1 has no valid architecture, ignoring.
Package libavahi-dbus-support version 0.8-1 has no valid architecture, ignoring.
Package libavahi-nodbus-support version 0.8-1 has no valid architecture, ignoring.
Package libcap version 2.27-1 has no valid architecture, ignoring.
Package libcares version 1.15.0-4 has no valid architecture, ignoring.
Package libdaemon version 0.14-5 has no valid architecture, ignoring.
Package libdb47 version 4.7.25.4.NC-5 has no valid architecture, ignoring.
Package libdbus version 1.12.12-1 has no valid architecture, ignoring.
Package libev version 4.25-1 has no valid architecture, ignoring.
Package libevdev version 1.6.0-1 has no valid architecture, ignoring.
Package libexpat version 2.2.9-1 has no valid architecture, ignoring.
Package libfaad2 version 2.9.2-1 has no valid architecture, ignoring.
Package libffmpeg-audio-dec version 3.4.7-3 has no valid architecture, ignoring.
Package libffmpeg-full version 3.4.7-3 has no valid architecture, ignoring.
Package libffmpeg-mini version 3.4.7-3 has no valid architecture, ignoring.
[... 143 lines cut ...]
Package python-jdcal version 1.4.1-1 has no valid architecture, ignoring.
Package python3-asyncio version 3.7.9-1 has no valid architecture, ignoring.
Package python3-base version 3.7.9-1 has no valid architecture, ignoring.
Package python3-codecs version 3.7.9-1 has no valid architecture, ignoring.
Package python3-email version 3.7.9-1 has no valid architecture, ignoring.
Package python3-light version 3.7.9-1 has no valid architecture, ignoring.
Package python3-logging version 3.7.9-1 has no valid architecture, ignoring.
Package python3-openssl version 3.7.9-1 has no valid architecture, ignoring.
Package python3-pytz version 2019.3-1 has no valid architecture, ignoring.
Package python3-urllib version 3.7.9-1 has no valid architecture, ignoring.
Package python3-xml version 3.7.9-1 has no valid architecture, ignoring.
Package python3 version 3.7.9-1 has no valid architecture, ignoring.
Package python version 2.7.18-3 has no valid architecture, ignoring.
Package sbc version 1.4-2 has no valid architecture, ignoring.
Package shine version 3.1.1-1 has no valid architecture, ignoring.
Package strongswan version 5.8.2-2 has no valid architecture, ignoring.
Package yajl version 2.1.0-2 has no valid architecture, ignoring.
Unknown package 'python3'.
Collected errors:
 * opkg_install_cmd: Cannot install package python3.
root@OpenWrt:~# 

```

疯狂提示`has no valid architecture, ignoring.`,我好懵啊。。。

cnjn, 2020-11-15:
目测python3的ipk包在`openwrt_packages2 https://op.supes.top/19.07.4/packages/mips_24kc/packages
`里面

garypang13, 2020-11-17:
更新到最新版固件即可解决,

cnjn, 2020-11-17:
> 更新到最新版固件即可解决,

成功解决问题！谢谢

closed 2020-11-17 by cnjn