kiddin9/Kwrt #189: 大佬，K2P的UPNP用不了啊……

ywyha, 2020-11-17:
刷了最近两个版本的K2P 16M固件，都无法正常UPNP……

garypang13, 2020-11-17:
UPNP是玄学...

ywyha, 2020-11-17:
> UPNP是玄学...

大佬，麻烦把玄学变成科学@@，我喜欢这个固件，或者大佬能不能告诉我旧版的哪个固件UPNP大概正常，我也是今天第一次用大佬的固件，还有不知道装哪个软件包能有KMS服务器

ywyha, 2020-11-17:
[https://i.loli.net/2020/11/17/CLoFc7ly1v5qbVP.png](url)

喜欢这个固件，DNS等很好用，如果UPNP正常的话，那真的是太好了，感谢大佬的付出，牛掰~

garypang13, 2020-11-17:
[image]

我用迅雷测试 UPNP是正常的.

ywyha, 2020-11-17:
> [image]
> 
> 我用迅雷测试 UPNP是正常的.

大佬，我知道你的UPNP好使，就是我纳闷我的UPNP为什么不好使……不知道是不是初始化设置的过程中触发了程序BUG。刷了三遍都没好使，晕菜了

ywyha, 2020-11-17:
[image]

折腾了半天，刷了大佬最近的两个版本，和发布列表最早的版本，依然不行，为了排除其他外在因素，我又重新刷回了lean大佬的潘多拉 R20.3.8 for K2P版本，upnp立马又正常了……而且我的局域网内不止一个UPNP，用lean大佬的全部正常，用garypang大佬的就不成功，每次刷完我都恢复出厂设置一次，尝试了三个版本，依然不行，还有大佬，最新的$$版本怎么不支持插件了呢……

garypang13, 2020-11-17:
下一版我换个upnp版本试试, k2p的ssr如果需要使用v2ray的话,需要在软件包中自行安装.

ywyha, 2020-11-17:
> 下一版我换个upnp版本试试, k2p的ssr如果需要使用v2ray的话,需要在软件包中自行安装.

大佬也熬夜吗，我刚又从lean的潘多拉刷回了11.12.2020版本，UPNP又意（悲）料（催）之（的）中的失败了，233333333

以前也碰到过固件UPNP失效的情况，就是这个版本PandoraBox 19.02 2019-02-01 https://www.right.com.cn/forum/thread-2651892-1-1.html

当时PandoraBox官方论坛有人反馈，然后PandoraBox坛友给出的解决办法是安装旧版，现在原帖我已经找不到了，所以也不知道是哪个版本的UPNP了，因为官方论坛貌似关闭了……

感觉是UPNP版本的问题，设置中无论是IGDv1还是IGDv2都无法正常使用UPNP，而且我局域网内好几个设备都需要UPNP，但是全都无法安装成功，刚才同环境下刷lean版本的潘多拉 R20.3.8 for K2P就全部正常。

还有大佬，我不是使用$$R，我想使用$$ NEW的obfs插件，可是最新版这个选项不见了……怎么弄出来

ywyha, 2020-11-17:
相对于论坛整合一堆东西的固件来说，我好喜欢这个固件，没有杂七杂八的东西，如果有，那就是刚需的东西，没有的也可以自己安装，新版的DNS，刚才试了下，貌似很好用，被污染的域名也能正常访问了，好省心~~谢谢大佬，刚才从倒数第二版在线升级到最新版，尝试了好几次，都没法升级成功，最后breed刷了最新版。谢谢大佬~~

garypang13, 2020-11-18:
试下今天编译的新版,换了个upnp版本.

ywyha, 2020-11-18:
大佬，我刚到家，看到回帖后第一时间刷上了今天的新版，依然不行，局域网内的所有需要UPNP的设备没有一个成功的……
upnp这个东西，因为需求的用户数量并不多，所有功能感知度很低，记着以前官方版的PandoraBox 19.02 UPNP出问题的时候，论坛里几乎没有发帖反馈的。

刚才又看了下当时PandoraBox K2P UPNP出问题的时候后来的解决方式，当时官方论坛有人发帖，把miniupnpd2.1换回旧版的2.0问题解决，据说当时PandoraBox K2P 19.03也是采用替换旧版的方式解决了这个问题，我当时参照帖子换回旧版的miniupnpd，UPNP，确实正常了。

而且这两天我在刷大佬固件之前又刷了论坛的K2P LEDE版本，UPNP依然不正常，而且我发现了之前有过反馈lede UPNP不正常的帖子，而且问题依旧是新版的miniupnpd2.1 UPNP不正常，https://koolshare.cn/thread-175389-1-1.html

期间也看过其他类似的帖子https://www.right.com.cn/forum/thread-393007-1-1.html，貌似UPNP不正常的固件都是使用了miniupnpd2.1

所以，大佬，我觉得问题的元凶，就是miniupnpd2.1这个锅，能不能换回2.0版本，小白不会弄，对UPNP有极大的需求，我快冒烟了，大佬求救2333333333333333
==========================================================
下边图片是当时PandoraBox官方论坛发的帖子，论坛挂了，貌似软件源还能用……
[image]

garypang13, 2020-11-18:
我的上一版用的就是老版的miniupnpd2.0, 新版才换的miniupnpd2.1, 不知道你用的是哪些需要upnp的软件来做的测试,我测试了迅雷和bitcomet比特彗星这两个软件,upnp都是正常的.
[image]

ywyha, 2020-11-18:
大佬 我用的是迅雷11 、qbittorrent，还有UPnPTest工具，还有局域网内的群辉、威联通内的P2P工具还有局域网内其他支持UPNP的设备都无法正常UPNP，而且我刚才又同环境下反复刷了lean R20.3.8 for K2P，再切回来，每次刷完都重置一次系统。而且还断开了局域网内其他设备的网络连接，依然无法UPNP成功，换回之前lean R20.3.8 for K2P又立马UPNP成功。刚才刷lean固件查看了下UPNP版本，我不知道是否跟小版本有关系， 我多希望是偶然性因素，可是我想不出任何我环境因素和其他人的差别……只连接一台有线PC，其他的设备电源我都断开了，依然无法成功UPNP……
[image]

garypang13, 2020-11-19:
你的这个 lean R20.3.8 for K2P 是在哪里下载的, 我的老版本miniupnpd是 2.0.20170421版, 看到你说刷了论坛的K2P LEDE版本，UPNP依然不正常,我试试换你用的20180503版试试,

ywyha, 2020-11-19:
大佬，在这个帖子下载的，我喜欢大佬这个固件，但是UPNP快让我神经衰弱了 233333333
 

https://www.right.com.cn/forum/thread-3135031-1-1.html

链接: https://pan.baidu.com/s/1stxwz00OrVrTgG554l2fBQ 提取码: hiif

ywyha, 2020-11-19:
还有去年的PandoraBox 官方的K2P 19.02版本 UPNP也是不正常，后来我用官方论坛网友发的方法替换了下边软件源的UPNP，upnp也恢复了正常，就是用的下边的PandoraBox的软件源。pangubox论坛貌似挂了，所以原帖找不到了……但是貌似软件源还活着……

http://downloads.pangubox.com:6380/pandorabox/18.10/packages/mipsel_1004kc_dsp/packages/
http://downloads.pangubox.com:6380/pandorabox/18.10/packages/mipsel_1004kc_dsp/luci/

garypang13, 2020-11-19:
新编了个带upnp 2.0.20200503版的固件,你再测试下

ywyha, 2020-11-19:
大佬，刚到家吃完饭，第一时间刷上了最新的固件，泪流满面，UPNP出现了，并且UPnPTest测试显示正常。
我为了避免是偶然性因素导致的成功，我把固件刷成了之前的版本，结果发现UPnPTest死活不通过，但是部分软件成功，不是全部成功，这两天被UPNP折腾的时候搜寻了一些帖子，记得有说过迅雷和一些软件的UPNP跟其他的不一样。UPnPTest一直显示failed，现在觉得之前的UPNP版本，并不是完全会失败，而是会部分成功，貌似兼容性不太好？或者我初始化的时候触发了BUG？

刚刷回旧版UPNPTEST显示失败后，又刷回了最新版固件，UPNPTEST测试立马显示passed，暂时没发现UPNP失败的设备。

我不知道新版UPNP是否加入了提升性能的新特性，如果固件能出稳定版和测试版就好了。

总之，感谢大佬，UPNP终于正常了。

garypang13, 2020-11-19:
好的,感谢反馈,那以后就默认这个版本的upnp了.

ywyha, 2020-11-19:
这个固件的界面和DNS功能都很好用，赞，捆绑一堆杂七杂八的固件中，一股清流。谢谢大佬！

ywyha, 2020-11-19:
大佬，自从这两天用这个固件一直在折腾UPNP，然后在尝试各种版本的时候中途出现过一次，本地网卡断开两次，我不记得当时使用的是哪个固件了，之前头两年我刷其他固件也出现过本地网卡断开的情况，当时是因为超频，我看恩山帖子中有CPU 频率 1100的帖子，我不确定是否刷了1100频率的固件，不知道是否是因为频率引起的，毕竟CPU频率体质不一样，超频到1100，我觉得还是太激进了，而且我路由器还有主动散热底座。貌似最新固件是1000频率的，今天下午刷到现在还没有出现过同样的网卡断开的情况。
===================================================================================
然后大佬，这两天我在反复折腾最近几版固件的时候，偶尔上手机游戏，使用5GWIFI连接频，繁出现延迟460或者异常卡的情况，这种情况，我之前从来没有出现过，今晚的最新固件我尝试重刷一次，依然没有解决问题，更纳闷的是在使用5GWIFI出现异常游戏延迟的时候关掉WIFI，尝试使用数据网络，延迟依然会异常，貌似手机会被污染一段时间。然后我又反复折腾，我关掉了chinadns，采用gfw模式，然后游戏延迟恢复了正常，刚才连续两局没有出现过异常情况。刚用大佬固件两天，而且一直在折腾UPNP，之前还真没有心思关心问题是否是由于固件引起的，但是今天晚上UPNP的问题解决了，所以今天又跑去玩手游了，然后发现问题很严重@@,所以又开始折腾，之前开着chinadns的时候貌似没有一局游戏是正常的，我今天明天再看看关掉chinadns是否还会再次出现同样情况，再来反馈，

garypang13, 2020-11-19:
@ywyha 修复了下,在后台软件包中刷新软路列表,在线更新下ssr-plus再试试,

ywyha, 2020-11-19:
@garypang13 
谢谢大佬，我去试试~

ywyha, 2020-11-19:
@garypang13 大佬，安装完1.2-21版本，重启了一次路由器后依然不行只要使用chinadns，WIFI下进游戏立马延迟异常，切回gfw，立马正常。
而且chinadns模式有很多未被污染的域名被强制代理了，导致本身访问速度飞快的网站绕路，比如rrys2020.com，gfw模式会直接访问，速度很快，一旦开启chinadns就会绕路变慢了，很多没被污染的网站都是这样子
[image]
[image]

garypang13, 2020-11-19:
游戏延迟应该是和你的路由器性能不足有关,关闭游戏模式UDP节点试试.

chinadns模式就是所有国外的网站全部走代理, gfw模式才是只代理被墙的网站,
以rrys2020.com这个网站为例,因为你是美国的节点,所以通过代理解析到离你节点最近的IP肯定也是美国的, ping是不会走代理的, 就好比你能打开google.com 但你无法ping通google.com 一样, 你ping的话是直接从你本地访问这个美国IP,没有经过代理,所以延迟肯定大了.  而实际你访问的时候你是经过代理到美国,从美国访问这个美国IP,所以速度并不慢的. 
gfw模式的话就是直接解析到了离你本地最近的香港IP,然后直接本地连接这个香港IP,延迟当然小了,

ywyha, 2020-11-19:
@garypang13 感谢大佬解释的这么详细，我原以为chinadns那个模式是防止DNS污染的，我以为没被污染的不走代理……看来我理解错意思了。（小白的无奈……@@）

至于WIFI下游戏延迟，我感觉更像是当时使用chinadns模式后游戏被代理了……，因为PC的游戏暂时是正常的，那个游戏模式UDP开关，我并没有开启。

刚才我尝试使用chinadns模式进入游戏，这回在游戏大厅内，WIFI延迟显示正常，之前异常的时候延迟都是200多，跟我节点的延迟貌似相匹配，所以我感觉更像是异常的时候被节点代理了。

我再尝试使用chinadns模式试试……大佬早点休息，感谢啦

garypang13, 2020-11-19:
@ywyha 游戏都是通过TCP IP直连的,如果没开TCP节点的话,是不会被代理的,可能就是你的网络波动引起的.

ywyha, 2020-11-19:
@garypang13 大佬，我也在尽可能的排除其他因素引起的。
在最近两三天刷大佬的固件之前，网络从来没有这么波动过，没有一次游戏延迟，掉线，最近半年从来没有过一次。

···刷大佬的固件后从头两天的默认开启chinadns模式开始，这两天玩手机游戏(王者农药)必然出现延迟异常的情况，从进入游戏大厅开始延迟就异常。是严重异常，不像是波动，因为延迟虽然高，但是相对稳定。

···昨晚解决UPNP问题后，我才有心思玩手机游戏，并且持续异常，开启gfw模式后，恢复正常，并且今晚开启gfw模式后再也没有出现异常过，但是一旦开启chinadns模式，就又异常了

···PC游戏开启chinadns模式，这两天我没发现延迟异常，手机游戏我只玩王者农药，所以别的游戏是否异常我不知道……

···最近版本的$$的PAC异常，会异常代理局域网内的IP，我当时折腾了好几天才发现是飞机的问题……所以我不知道我这个延迟异常会不会是类似的问题。

···刚才我尝试开启chinadns模式，一进游戏，又异常了……

···这几天因为UPNP反复折腾大佬的各个版本，貌似大佬是最近两个版本才改成默认chinadns模式的，之前貌似都是默认gfw模式，所以chinadns模式是否完全正常，用户的感知度很低……就像UPNP一样

···然后，我最近两天我会一直尝试切换模式，看看是否与开关chinadns模式有关联，但是这两天我已经尝试切换模式10多次了，都是100%符合开chinadns模式会延迟异常，开gfw模式从来不会异常的套路，无一例外。

···作为小白，我只能用这种切换模式的笨方法试了，因为其他的我也不懂……近几天我会一直留意，延迟异常是否跟开启chinadns模式有绝对关系，所以大佬，暂时先不用特殊对待。（额，今晚我才知道chinadns模式是绕过大陆IP，我一般也不用这个模式，感觉太暴力了@@）

ywyha, 2020-11-20:
大佬，经过了一天时间，我玩了2个小时手机游戏，gfw模式，从没有出现延迟异常过。

刚才尝试再次开启chinadns模式，然后尝试进入手游第一局，又是全程延迟异常。

我昨天原以为是手游被节点代理了，但是今天发现延迟与节点延迟并不符，正常情况下游戏延迟50左右，我到节点的延迟需要240左右，但是今天游戏延迟异常的时候游戏延迟是180，我刚看了下节点，我到节点的最低延迟也需要230多。所以可能并不是被代理了。

但是，现在看十有八九延迟异常与开启chinadns模式有关，因为关闭后一天都没有异常过，结果开启后的第一局游戏又出现了延迟异常，不是正常的波动，因为全程都是延迟180以上。

腾讯的PC游戏我没发现异常延迟，异常延迟只出现在我手机WIFI下玩的王者荣耀游戏，其他游戏我没玩所以不知道。5GWIFI和2.4G都会异常，

garypang13, 2020-11-21:
经过测试,确实存在chinadns模式(大陆白名单)玩农药高延迟460的情况,
游戏会随机通过TCP连接某个不固定的香港IP,大陆白名单会通过代理连接这个IP,所以会被游戏认为是海外用户而走海外的线路
如果要玩这个游戏的话请使用gfwlist模式(不清楚其他腾讯游戏是否也有这个情况)

ywyha, 2020-11-21:
@garypang13 大佬，我平时也不用这个模式，我只是这几天刚才开始用大佬的固件，然后一直折腾UPNP问题，当时根本无心关心其他，因为不用这个模式，所以对我倒是没有影响，就是不知道其他用户用的人多不多。

DNS污染对我真的很头疼，用所谓的防污染DNS确实解决了部分域名被污染的问题，但是对国内网站解析不准导致访问速度变慢，用绕过大陆IP方式导致本身没有被污染的网站访问速度变慢，如果有好的方法就好了，哈哈~

garypang13, 2020-11-21:
固件里的ssr不管哪种模式都不会存在DNS污染问题的.
用绕过大陆IP方式导致本身没有被污染的网站访问速度变慢,这个是不成立的,代理不仅仅是用于翻墙,最重要的是国外网站加速,国外网站即使没有被墙,你如果从国内直连到国外比如美国,速度是很慢的,特别是下载的时候经常只有几kb/s.而大陆白名单模式只要你的节点速度好,所有国外网站打开速度都会有质的飞跃.
rrys2020.com这个网站之所以直连速度还不错是因为在香港有节点,国内直连香港所以速度不会慢.

ywyha, 2020-11-21:
@garypang13 大佬，这个域名api.themoviedb.org，我尝试了下，并没有解析到能连通的地址。 

刷完21日固件，然后用了一会出现ShadowSocksR Plus+ 客户端未运行，尝试关闭重新开启，失败了，然后重启路由器后，ShadowSocksR Plus+提示运行中，但是出现了过一会出现了电脑无法访问google的情况，查看路由器内的状态【谷歌】连通性检查提示失败，然后这个时候尝试用PC端的SS客户端连接同节点，提示连接正常，路由器内的同节点，依然异常。

短短一个小时内出现过两次问题，感觉并不像是偶然性发生的，而且我是刚才全新刷的21日固件。

ywyha, 2020-11-23:
大佬，我还想问下，怎样才能添加对trojan-go的支持，我安装软件包内的trojan-go，ssr内也没有出现trojan-go的选项呀……@@ 哪里弄得不对……

ywyha, 2020-11-24:
[image]

大佬，刷了23日固件，固件中的一些功能是越来越符合使用习惯了。像之前默认带去广告还真用不到，这次也去掉了。

就是点击软件包中的列表会提示好几十个软件包可更新，然后点击更新列表还出错。

closed 2020-11-24 by ywyha