kiddin9/Kwrt #190: 红米ac2100里面的python3是一个混血版

cnjn, 2020-11-17:
我在后台安装了python3

这是运行界面：

```bash
root@OpenWrt:~# python3
Python 3.9.0 (default, Nov 15 2020, 17:48:33) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

随后我又安装了python3-pip

然后就报错了：

```bash
root@OpenWrt:~# pip3
Traceback (most recent call last):
  File "/usr/bin/pip3", line 5, in <module>
    from pip._internal.cli.main import main
ModuleNotFoundError: No module named 'pip'
root@OpenWrt:~# 
```

经过多方查找，我找到了问题所在：我安装了python3.9，却装了个pip3.7。。。

如下所示：

```bash
root@OpenWrt:~# python
python     python3    python3.9
root@OpenWrt:~# pip3
pip3    pip3.7

```

诡异的是，我安装的时候是没得选的，只能安装3.9版本，而且在安装3.9的同时居然夹带着一部分3.7。。。

以下可以证明：

```bash
root@OpenWrt:~# ls /usr/lib/python3.
python3.7/  python3.9/
```

我安装python3，路由器中居然同时出现了python3.7和python3.9的文件夹，我卸载python3，则python3.7和python3.9的文件夹同时消失。也就是说，我安装了一个混血版python？心态崩了啊。。。这导致pip3根本无法使用。

garypang13, 2020-11-17:
把 /etc/opkg/distfeeds.conf中的 这条 `src/gz openwrt_packages https://op.supes.top/snapshots/packages/x86_64/packages` 删掉,更新软件列表,再重装python3

cnjn, 2020-11-18:
> 把 /etc/opkg/distfeeds.conf中的 这条 `src/gz openwrt_packages https://op.supes.top/snapshots/packages/x86_64/packages` 删掉,更新软件列表,再重装python3

我的opkg配置中没有这条啊。。。

```bash
root@OpenWrt:~# cat /etc/opkg/distfeeds.conf 
src/gz openwrt_core https://op.supes.top/snapshots/targets/ramips/mt7621/packages
src/gz openwrt_base https://op.supes.top/snapshots/packages/mipsel_24kc/base
src/gz openwrt_freifunk https://op.supes.top/snapshots/packages/mipsel_24kc/freifunk
src/gz openwrt_luci https://op.supes.top/snapshots/packages/mipsel_24kc/luci
src/gz openwrt_packages https://op.supes.top/snapshots/packages/mipsel_24kc/packages
src/gz openwrt_routing https://op.supes.top/snapshots/packages/mipsel_24kc/routing
src/gz openwrt_telephony https://op.supes.top/snapshots/packages/mipsel_24kc/telephony
root@OpenWrt:~# cat /etc/opkg/customfeeds.conf 
# add your custom package feeds here
#
# src/gz example_feed_name http://www.example.com/path/to/files
src/gz openwrt_custom https://op.supes.top/packages/mipsel
src/gz openwrt_packages2 https://op.supes.top/19.07.4/packages/mipsel_24kc/packages
src/gz openwrt_base2 https://op.supes.top/19.07.4/packages/mipsel_24kc/base

```

garypang13, 2020-11-18:
@cnjn 修复了,更新软件列表后重新安装

cnjn, 2020-11-18:
@garypang13 这一次的pip能安装了，但是无法正常使用。。。
```bash
root@OpenWrt:~# pip

Usage:   
  pip <command> [options]

Commands:
  install                     Install packages.
  download                    Download packages.
  uninstall                   Uninstall packages.
  freeze                      Output installed packages in requirements format.
  list                        List installed packages.
  show                        Show information about installed packages.
  check                       Verify installed packages have compatible dependencies.
  config                      Manage local and global configuration.
  search                      Search PyPI for packages.
  cache                       Inspect and manage pip's wheel cache.
  wheel                       Build wheels from your requirements.
  hash                        Compute hashes of package archives.
  completion                  A helper command used for command completion.
  debug                       Show information useful for debugging.
  help                        Show help for commands.

General Options:
  -h, --help                  Show help.
  --isolated                  Run pip in an isolated mode, ignoring environment variables and user configuration.
  -v, --verbose               Give more output. Option is additive, and can be used up to 3 times.
  -V, --version               Show version and exit.
  -q, --quiet                 Give less output. Option is additive, and can be used up to 3 times (corresponding to WARNING, ERROR, and CRITICAL logging levels).
  --log <path>                Path to a verbose appending log.
  --no-input                  Disable prompting for input.
  --proxy <proxy>             Specify a proxy in the form [user:passwd@]proxy.server:port.
  --retries <retries>         Maximum number of retries each connection should attempt (default 5 times).
  --timeout <sec>             Set the socket timeout (default 15 seconds).
  --exists-action <action>    Default action when a path already exists: (s)witch, (i)gnore, (w)ipe, (b)ackup, (a)bort.
  --trusted-host <hostname>   Mark this host or host:port pair as trusted, even though it does not have valid or any HTTPS.
  --cert <path>               Path to alternate CA bundle.
  --client-cert <path>        Path to SSL client certificate, a single file containing the private key and the certificate in PEM format.
  --cache-dir <dir>           Store the cache data in <dir>.
  --no-cache-dir              Disable the cache.
  --disable-pip-version-check
                              Don't periodically check PyPI to determine whether a new version of pip is available for download. Implied with --no-index.
  --no-color                  Suppress colored output
  --no-python-version-warning
                              Silence deprecation warnings for upcoming unsupported Pythons.
  --use-feature <feature>     Enable new functionality, that may be backward incompatible.
  --use-deprecated <feature>  Enable deprecated functionality, that will be removed in the future.
root@OpenWrt:~# pip list
Traceback (most recent call last):
  File "/usr/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/pip/_internal/cli/main.py", line 73, in main
  File "/usr/lib/python3.9/site-packages/pip/_internal/commands/__init__.py", line 104, in create_command
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3.9/site-packages/pip/_internal/commands/list.py", line 9, in <module>
  File "/usr/lib/python3.9/site-packages/pip/_internal/cli/req_command.py", line 16, in <module>
  File "/usr/lib/python3.9/site-packages/pip/_internal/index/collector.py", line 5, in <module>
ModuleNotFoundError: No module named 'cgi'
root@OpenWrt:~#
```

garypang13, 2020-11-18:
[image]

再重装次

cnjn, 2020-11-19:
@garypang13 谢谢，我重装成功了！
话说用这个路由器来安装模块真是提心提心吊胆，内存占用直接100%，128M用到只剩100k:laughing:

closed 2020-11-19 by cnjn

garypang13, 2020-11-19:
后台那个内存显示不准的,在ssh中用htop命令来查看实际内存占用.

cnjn, 2020-11-19:
> 后台那个内存显示不准的,在ssh中用htop命令来查看实际内存占用.

不不不，我第一次安装requests的时候pip进程真的被系统kill了一次:joy:
重启后才安装好:grin: