kiddin9/Kwrt #197: 大佬，23号版的K2P $$R固件和redmi AC2100$$R固件不一样是升级什么了吗

ywyha, 2020-11-24:
我用K2P的23号固件好像没有污染，之前的固件无法ping通的域名也ping的通，貌似域名解析速度也快了一些。

用23号版的AC2100固件，K2P能连接的域名AC2100依然无法连接……

ywyha, 2020-11-24:
Adblock Plus+ 好不想不稳定，刚才AC2100关了Adblock Plus+之后电脑和手机都连不上网了……重启路由器才恢复，固件是今天全新刷的

garypang13, 2020-11-24:
Adblock Plus问题已修复,在软件包中更新此插件即可
我解释过了ping是没用的,你无论如何Ping不同google,但不影响你访问google,要测试也应该用nslookup 命令

ywyha, 2020-11-24:
@garypang13 大佬，我使用K2P 23号固件确实api.themoviedb.org这个域名能ping的通……对应的服务也能正常使用。
之前的固件不可以。
AC2100最新的固件也不可以。
我的意思是23号的K2P固件这个被污染的域名api.themoviedb.org返回了国内能连接的IP，而之前的K2P固件和23号的AC2100固件都不行，我的意思是是不是23号的K2P固件升级了什么，还是偶然性的解析到了国内能连接的地址……因为之前的固件使用这个API一直都是连接失败，唯独23号的K2P连接成功了，而且$$R界面也变了
[image]

garypang13, 2020-11-24:
api.themoviedb.org这个网站是被墙了的,但是gfwlist里并没有收录这个网站,所以要想访问这个网站就只能使用智能模式或者手动将这个域名加到强制走代理的域名中

ywyha, 2020-11-24:
> api.themoviedb.org这个网站是被墙了的,但是gfwlist里并没有收录这个网站,所以要想访问这个网站就只能使用智能模式或者手动将这个域名加到强制走代理的域名中
大佬，这就是我奇怪的地方，我纳闷的是为什么我刷了23号的K2P固件使用gfw模式却突然能解析出了没有被污染的IP……我用的并不是智能模式，我之前刷的大佬的K2P固件使用gfw模式一直都无法连接这个域名。23号的却突然可以…而且23号固件解析出来的IP跟之前的也不一样…而且23号的K2P固件我感觉解析速度变快，不知道是不是幻觉……

这个域名是被污染了，并不是完全被墙的……

我看23号的K2P固件SSR界面变了，我不知道和这个有没有关系……

garypang13, 2020-11-24:
SSR界面都是一样的

ywyha, 2020-11-24:
> SSR界面都是一样的

嗯，谢谢大佬了。大佬说一样我也没办法，反正我23号的K2P的$$R界面跟之前的不一样了，$$R里边增加了去广告开关，然后服务软件包里没了Adblock Plus+和uHTTPd，界面看着更简洁了。 ACC网络加速里增加了CND加速。挺喜欢这个版本的，不知道AC2100什么时候也能用上同样的界面

garypang13, 2020-11-24:
> > SSR界面都是一样的
> 
> 嗯，谢谢大佬了。大佬说一样我也没办法，反正我23号的K2P的$$R界面跟之前的不一样了，$$R里边增加了去广告开关，然后服务软件包里没了Adblock Plus+和uHTTPd，界面看着更简洁了。 ACC网络加速里增加了CND加速。挺喜欢这个版本的，不知道AC2100什么时候也能用上同样的界面

那一定是你刷错固件了.

ywyha, 2020-11-24:
@garypang13 大佬……尴尬了，确实“刷错了”，刷成了9月23号的固件……
不过大佬我又试了下9月23号的固件的DNS解析好像确实比新版的要快……而且这个被污染的域名解析到了国内能连接的正确的地址api.themoviedb.org
[image]

而新版的K2P固件就不行，无法解析到正确地址……解析成了157.240.2.36被墙的IP

而且9月23号的固件解析网站感觉确实更快一些……新版的有时候打开腾讯的部分网站都要卡那么一下

还有大佬，希望能增加防污染的开关…

ywyha, 2020-11-24:
大佬，其实想增加防污染开关主要是，我访问一些国内网站的速度会变慢，，导致访问国内的很多速度变慢。我用手机的数据网络访问腾讯会解析到本省的服务器……访问速度差别大大滴……

所以大佬，希望能增加个防污染开关。运营商DNS虽然污染比较头疼，但是一般都会解析到距离自己最近的服务器……对我这里来说访问网站的速度区别还是挺明显的……我现在解析到的腾讯首页网站服务器距离我几千公里……其他很多网站都是无法解析到最近服务器……

closed 2020-11-27 by ywyha