kiddin9/Kwrt #198: K2P 电信pppoe拨号不成功

jackadam1981, 2020-11-24:
主机名	OpenWrt
型号	Phicomm K2P
架构	MediaTek MT7621 ver:1 eco:3
固件版本	OpenWrt SNAPSHOT 11.21.2020 by GaryPang / LuCI Master git-4dbf600
内核版本	5.4.77

出现这几个错误
错误: 未知错误（USER_REQUEST）
错误: 尝试连接失败
错误: 未知错误（NEGOTIATION_FAILED）

相关日志如下：
Sun Nov 22 00:08:03 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 22 00:08:03 2020 daemon.notice netifd: Interface 'wan' is disabled
Sun Nov 22 00:08:03 2020 daemon.notice netifd: Interface 'wan' is enabled
Sun Nov 22 00:08:03 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 22 00:08:03 2020 daemon.err insmod: module is already loaded - slhc
Sun Nov 22 00:08:03 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 22 00:08:03 2020 daemon.err insmod: module is already loaded - pppox
Sun Nov 22 00:08:03 2020 daemon.err insmod: module is already loaded - pppoe
Sun Nov 22 00:08:03 2020 daemon.notice netifd: wan (11471): uci: Entry not found
Sun Nov 22 00:08:03 2020 daemon.notice netifd: wan (11471): sh: out of range
Sun Nov 22 00:08:03 2020 daemon.info pppd[11528]: Plugin rp-pppoe.so loaded.
Sun Nov 22 00:08:03 2020 daemon.info pppd[11528]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Nov 22 00:08:03 2020 daemon.notice pppd[11528]: pppd 2.4.8 started by root, uid 0
Sun Nov 22 00:08:03 2020 daemon.info pppd[11528]: PPP session is 12941
Sun Nov 22 00:08:03 2020 daemon.warn pppd[11528]: Connected to 00:25:12:52:16:33 via interface eth0.2
Sun Nov 22 00:08:04 2020 daemon.info pppd[11528]: Renamed interface ppp0 to pppoe-wan
Sun Nov 22 00:08:04 2020 daemon.info pppd[11528]: Using interface pppoe-wan
Sun Nov 22 00:08:04 2020 daemon.notice pppd[11528]: Connect: pppoe-wan <--> eth0.2
Sun Nov 22 00:08:34 2020 daemon.warn pppd[11528]: LCP: timeout sending Config-Requests
Sun Nov 22 00:08:34 2020 daemon.notice pppd[11528]: Connection terminated.
Sun Nov 22 00:08:34 2020 daemon.info pppd[11528]: Sent PADT
Sun Nov 22 00:08:34 2020 daemon.notice pppd[11528]: Modem hangup
Sun Nov 22 00:08:34 2020 daemon.info pppd[11528]: Exit.

Sun Nov 22 00:09:35 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 22 00:09:36 2020 daemon.notice netifd: Interface 'wan' is disabled
Sun Nov 22 00:09:36 2020 daemon.notice netifd: Interface 'wan' is enabled
Sun Nov 22 00:09:36 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 22 00:09:36 2020 daemon.err insmod: module is already loaded - slhc
Sun Nov 22 00:09:36 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 22 00:09:36 2020 daemon.err insmod: module is already loaded - pppox
Sun Nov 22 00:09:36 2020 daemon.err insmod: module is already loaded - pppoe
Sun Nov 22 00:09:36 2020 daemon.notice netifd: wan (12551): uci: Entry not found
Sun Nov 22 00:09:36 2020 daemon.notice netifd: wan (12551): sh: out of range
Sun Nov 22 00:09:36 2020 daemon.info pppd[12608]: Plugin rp-pppoe.so loaded.
Sun Nov 22 00:09:36 2020 daemon.info pppd[12608]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Nov 22 00:09:36 2020 daemon.notice pppd[12608]: pppd 2.4.8 started by root, uid 0
Sun Nov 22 00:09:36 2020 daemon.info pppd[12608]: PPP session is 7289
Sun Nov 22 00:09:36 2020 daemon.warn pppd[12608]: Connected to 00:25:12:52:16:33 via interface eth0.2
Sun Nov 22 00:09:36 2020 daemon.info pppd[12608]: Renamed interface ppp0 to pppoe-wan
Sun Nov 22 00:09:36 2020 daemon.info pppd[12608]: Using interface pppoe-wan
Sun Nov 22 00:09:36 2020 daemon.notice pppd[12608]: Connect: pppoe-wan <--> eth0.2
Sun Nov 22 00:10:06 2020 daemon.warn pppd[12608]: LCP: timeout sending Config-Requests
Sun Nov 22 00:10:06 2020 daemon.notice pppd[12608]: Connection terminated.
Sun Nov 22 00:10:06 2020 daemon.info pppd[12608]: Sent PADT
Sun Nov 22 00:10:06 2020 daemon.notice pppd[12608]: Modem hangup
Sun Nov 22 00:10:06 2020 daemon.info pppd[12608]: Exit.

另一个拨号正常的日志：
Jan  1 00:00:15 pppd[397]: Plugin rp-pppoe.so loaded.
Jan  1 00:00:15 pppd[397]: RP-PPPoE plugin version 3.12 compiled against pppd 2.4.7
Jan  1 00:00:15 pppd[433]: pppd 2.4.7 started by admin, uid 0
Jan  1 00:00:15 crond[439]: crond (busybox 1.24.2) started, log level 8
Jan  1 00:00:15 pppd[433]: PPP session is 8491 (0x212b)
Jan  1 00:00:15 pppd[433]: Connected to 00:25:12:52:16:33 via interface eth2.2
Jan  1 00:00:15 pppd[433]: Using interface ppp0
Jan  1 00:00:15 pppd[433]: Connect: ppp0 <--> eth2.2
Jan  1 00:00:15 httpd[442]: Server listening port 80 (HTTP).
Jan  1 00:00:15 miniupnpd[446]: version 2.0 starting UPnP-IGD ext if ppp0 BOOTID=1420041615
Jan  1 00:00:15 miniupnpd[446]: HTTP listening on port 26752
Jan  1 00:00:15 kernel: Load Ralink WDG Timer Module
Jan  1 00:00:15 kernel: Netfilter messages via NETLINK v0.30.
Jan  1 00:00:16 kernel: ip_set: protocol 6
Jan  1 00:00:16 kernel: br0: port 3(rai0) entered forwarding state
Jan  1 00:00:16 kernel: br0: port 2(ra0) entered forwarding state
Jan  1 00:00:16 kernel: br0: port 1(eth2.1) entered forwarding state
Jan  1 00:00:20 pppd[433]: syncppp not active
Jan  1 00:00:20 pppd[433]: Remote message: Authentication success,Welcome!
Jan  1 00:00:20 pppd[433]: PAP authentication succeeded
Jan  1 00:00:20 pppd[433]: peer from calling number 00:25:12:52:16:33 authorized
Jan  1 00:00:20 pppd[433]: local  IP address 1.192.**
Jan  1 00:00:20 pppd[433]: remote IP address 1.192.**
Jan  1 00:00:20 pppd[433]: primary   DNS address 222.85.85.85
Jan  1 00:00:20 pppd[433]: secondary DNS address 222.88.88.88
Jan  1 00:00:20 RT-AC54U: WAN up (ppp0)

还有奇葩的是在另一条联通线路上拨号正常，换电信线路这里就拨号不正常。

ywyha, 2020-11-25:
我是AC2100，有时候重启WAN口会直接导致本地连接断开……过一会才恢复……

ywyha, 2020-11-25:
我刚又试了下AC2100 点击重启WAN口，炸的几率很高，有时候本地连接断开过一会才会恢复，有时候直接炸了，连路由器管理界面都进不去……几乎每重启两次就会炸一次……

jackadam1981, 2020-11-29:
Sun Nov 29 11:12:16 2020 daemon.err pppd[7576]: select (waitForPADS): Interrupted system call
Sun Nov 29 11:12:16 2020 daemon.warn pppd[7576]: Timeout waiting for PADS packets
Sun Nov 29 11:12:16 2020 daemon.err pppd[7576]: Unable to complete PPPoE Discovery
Sun Nov 29 11:12:16 2020 daemon.info pppd[7576]: Exit.
Sun Nov 29 11:12:16 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 29 11:12:16 2020 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sun Nov 29 11:12:16 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Nov 29 11:12:16 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 29 11:12:17 2020 daemon.err insmod: module is already loaded - slhc
Sun Nov 29 11:12:17 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 29 11:12:17 2020 daemon.err insmod: module is already loaded - pppox
Sun Nov 29 11:12:17 2020 daemon.err insmod: module is already loaded - pppoe
Sun Nov 29 11:12:17 2020 daemon.notice netifd: wan (7783): uci: Entry not found
Sun Nov 29 11:12:17 2020 daemon.notice netifd: wan (7783): sh: out of range
Sun Nov 29 11:12:17 2020 daemon.info pppd[7833]: Plugin rp-pppoe.so loaded.
Sun Nov 29 11:12:17 2020 daemon.info pppd[7833]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Nov 29 11:12:18 2020 daemon.notice pppd[7833]: pppd 2.4.8 started by root, uid 0
Sun Nov 29 11:12:33 2020 daemon.warn pppd[7833]: Timeout waiting for PADS packets
Sun Nov 29 11:12:33 2020 daemon.err pppd[7833]: Unable to complete PPPoE Discovery
Sun Nov 29 11:12:33 2020 daemon.info pppd[7833]: Exit.
Sun Nov 29 11:12:33 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 29 11:12:33 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 29 11:12:33 2020 daemon.err insmod: module is already loaded - slhc
Sun Nov 29 11:12:33 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 29 11:12:33 2020 daemon.err insmod: module is already loaded - pppox
Sun Nov 29 11:12:33 2020 daemon.err insmod: module is already loaded - pppoe
Sun Nov 29 11:12:33 2020 daemon.notice netifd: wan (7992): uci: Entry not found
Sun Nov 29 11:12:33 2020 daemon.notice netifd: wan (7992): sh: out of range
Sun Nov 29 11:12:33 2020 daemon.info pppd[8025]: Plugin rp-pppoe.so loaded.
Sun Nov 29 11:12:33 2020 daemon.info pppd[8025]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Nov 29 11:12:33 2020 daemon.notice pppd[8025]: pppd 2.4.8 started by root, uid 0
Sun Nov 29 11:12:48 2020 daemon.warn pppd[8025]: Timeout waiting for PADS packets
Sun Nov 29 11:12:48 2020 daemon.err pppd[8025]: Unable to complete PPPoE Discovery
Sun Nov 29 11:12:48 2020 daemon.info pppd[8025]: Exit.
Sun Nov 29 11:12:49 2020 daemon.notice netifd: Interface 'wan' is now down
Sun Nov 29 11:12:49 2020 daemon.notice netifd: Interface 'wan' is setting up now
Sun Nov 29 11:12:49 2020 daemon.err insmod: module is already loaded - slhc
Sun Nov 29 11:12:49 2020 daemon.err insmod: module is already loaded - ppp_generic
Sun Nov 29 11:12:49 2020 daemon.err insmod: module is already loaded - pppox
Sun Nov 29 11:12:49 2020 daemon.err insmod: module is already loaded - pppoe
Sun Nov 29 11:12:49 2020 daemon.notice netifd: wan (8159): uci: Entry not found
Sun Nov 29 11:12:49 2020 daemon.notice netifd: wan (8159): sh: out of range
Sun Nov 29 11:12:49 2020 daemon.info pppd[8192]: Plugin rp-pppoe.so loaded.
Sun Nov 29 11:12:49 2020 daemon.info pppd[8192]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Nov 29 11:12:49 2020 daemon.notice pppd[8192]: pppd 2.4.8 started by root, uid 0
Sun Nov 29 11:12:59 2020 daemon.info pppd[8192]: PPP session is 6519
Sun Nov 29 11:12:59 2020 daemon.warn pppd[8192]: Connected to d8:49:0b:b6:35:d6 via interface eth0.2
Sun Nov 29 11:12:59 2020 daemon.info pppd[8192]: Renamed interface ppp0 to pppoe-wan
Sun Nov 29 11:12:59 2020 daemon.info pppd[8192]: Using interface pppoe-wan
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: Connect: pppoe-wan <--> eth0.2
Sun Nov 29 11:12:59 2020 daemon.info pppd[8192]: syncppp not active
Sun Nov 29 11:12:59 2020 daemon.info pppd[8192]: CHAP authentication succeeded: Authentication success,Welcome!
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: CHAP authentication succeeded
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: peer from calling number D8:49:0B:B6:35:D6 authorized
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: local  LL address fe80::5955:ceb3:17de:b1c5
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: remote LL address fe80::da49:0bff:feb6:35d6
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: local  IP address 100.64.92.15
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: remote IP address 100.64.0.1
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: primary   DNS address 202.102.224.68
Sun Nov 29 11:12:59 2020 daemon.notice pppd[8192]: secondary DNS address 202.102.227.68
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Interface 'wan' is now up
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Interface 'wan_6' is enabled
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sun Nov 29 11:12:59 2020 daemon.notice netifd: Interface 'wan_6' is setting up now
Sun Nov 29 11:12:59 2020 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
[... cut ...]
Sun Nov 29 11:13:00 2020 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sun Nov 29 11:13:00 2020 daemon.warn odhcpd[1438]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Nov 29 11:13:00 2020 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Nov 29 11:13:05 2020 daemon.notice procd: /etc/rc.d/S99opkg: Downloading  https://op.supes.top/snapshots/targets/ramips/mt7621/packages/Packages.gz

再贴一份

jackadam1981, 2020-11-29:
SyntaxError
Failed to execute 'open' on 'XMLHttpRequest': Invalid URL

还经常出这个东西

jackadam1981, 2020-12-20:
刚试了最新的12.16-openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade.bin

还是拨号失败，改MTU1492也无效

closed 2021-01-07 by garypang13