kiddin9/Kwrt #200: ShadowSocksR Plus+ 设置里的smartdns不生效

jrokr, 2020-11-26:
nslookup时仍然返回多个地址。

garypang13, 2020-11-27:
默认关闭了smartdns的测速功能,

jrokr, 2020-11-27:
> 默认关闭了smartdns的测速功能,

这样不就无法返回最快的IP了吗？
难道测速有负面作用？

garypang13, 2020-11-27:
https://github.com/coolsnowwolf/lede/issues/2551#issuecomment-678796678

jrokr, 2020-11-27:
> [coolsnowwolf/lede#2551 (comment)](https://github.com/coolsnowwolf/lede/issues/2551#issuecomment-678796678)
了解了，不过如果可以最好能加个开关。

small-5, 2020-11-27:
在不考虑ECS协议的情况下，递归DNS出口与你是同省同ISP，测速没有任何意义，因为部署了CDN的网站的权威DNS会根据递归DNS的后端出口返回对于此IP最友好的一组IP。多IP是有冗余备灾的作用的，而且一般这些IP都是同一个地区甚至同一个机房的，延迟根本就不会有多大差距，有些可能就是几ms而已。所以在不考虑DNS劫持的情况下，使用ISP的DNS准确度才是最高的。折腾测速，不如去查查递归DNS的出口是不是你们当地同ISP的，如果公共DNS的后端出口是你们当地同ISP的IP，那解析准确度就和ISP的DNS没有区别。
https://nstool.netease.com/

jrokr, 2020-11-27:
> 在不考虑ECS协议的情况下，递归DNS出口与你是同省同ISP，测速没有任何意义，因为部署了CDN的网站的权威DNS会根据递归DNS的后端出口返回对于此IP最友好的一组IP。多IP是有冗余备灾的作用的，而且一般这些IP都是同一个地区甚至同一个机房的，延迟根本就不会有多大差距，有些可能就是几ms而已。所以在不考虑DNS劫持的情况下，使用ISP的DNS准确度才是最高的。折腾测速，不如去查查递归DNS的出口是不是你们当地同ISP的，如果公共DNS的后端出口是你们当地同ISP的IP，那解析准确度就和ISP的DNS没有区别。
> https://nstool.netease.com/

阿里出口是北京本地的，DNSPOD就到处都是了,DoH差到深圳，UPD就各地随机。那我现在就用阿里的了。
我测试的qq.com三个ip:5ms，26ms，42ms。

small-5, 2020-11-27:
dnspod就是这样的，出口经常是在广东的IP，但是因为有ECS协议，所以解析的准确度是可以的。。。5ms 26ms 42ms你能感觉出差别吗？你拿ISP的DNS解析出来的qq.com的IP应该也是同样的，不要把网站设计的多IP冗余备灾的功能给弄掉了，不然这个5ms的IP出现故障的时候，你是没有办法打开网站的。

small-5, 2020-11-27:
而且很多网站的主域名（包括你说的qq.com）是没有做CDN部署的，因为只是起一个跳转作用。你打开qq.com就跳到www.qq.com了。www.qq.com解析准确才是关键。

small-5, 2020-11-27:
比如taobao.com，你在任何地方任何ISP解析，目前的IP都是140.205.94.189与140.205.220.96，包括在海外解析，因为这个只是起跳转作用。www.taobao.com解析是不是CDN最优的才是关键。

jrokr, 2020-11-27:
> 比如taobao.com，你在任何地方任何ISP解析，目前的IP都是140.205.94.189与140.205.220.96，包括在海外解析，因为这个只是起跳转作用。www.taobao.com解析是不是CDN最优的才是关键。

了解了，谢谢。

closed 2020-12-01 by garypang13