kiddin9/Kwrt #201: 大佬，请问怎样更改CPU频率，我的小米AC2100路由刚才炸了……

ywyha, 2020-11-27:
大佬，头些天之前一直用lean大佬的固件，然后发现大佬的固件后刷了K2P 和AC2100，然后这些天K2P好像刷了一个9月的固件，出现过两次异常本地连接断开的情况，那个固件频率好像是1100，后来刷了新固件，暂时没发现炸的情况。

然后我的AC2100今天又刷了大佬最新的24号固件，全新刷入的，刚才出现了本地连接断开的情况，断开网络过一会自动恢复后，我赶紧登录路由器后台，看路由器运行时间，发现路由器自动重启过……

所以大佬，我路由器早期时候用breed曾经尝试过用breed超频，然后后来发现会降低稳定性，有时候会出现本地连接断开的异常情况，所以后来，我一直都是原频率运行……

头些年我真的喜欢超频，但是后来发现超频有些时候真的得不偿失，曾经用功率插座试过功耗的变化，发现最终还不如换平台了……

大佬，我不是要把你发布的固件改成默认频率（虽然我依旧认为超频对性能的影响微乎其微，降低潜在的稳定性，得不偿失），我只是想知道，我怎样才能修改路由器频率……或者大佬，你能不能在固件的界面中增加个设置频率的选项，大佬最近一周，我经历过三次，K2P和AC2100的本地连接断开，大佬，我现在非常抓狂@@，因为之前不超频的时候从来没有出现过这种情况。大佬，虽然我这两台路由器的CPU可能都是所谓的大雷。连1000hz都不稳定……但是既然我自己都一下子碰到两台这样，其他人估计也会碰到，毕竟是路由器，关系到网络稳定……刚才我正在上网，结果网络断开了，如果要是局域网内部网络传输设备刷机，这个时候断网，可能直接导致设备故障了，所以大佬，能不能想个办法应对我们这些CPU大雷的用户……而且我路由器还有主动散热底座，可是依然无法稳定……我自己就一连碰到两台大雷7621CPU，我觉得7621无法稳定运行在1000hz的概率真的不算很低……

ywyha, 2020-11-27:
大佬，救命，被动异常断网，无限抓狂&……

ywyha, 2020-11-27:
其实大佬，刷机几乎都要通过breed中进行，但是breed中有超频选项，所以大佬，固件中能不能增加个选择是否超频的选项……毕竟路由器稳定第一……

ywyha, 2020-11-27:
大佬，我现在都怀疑我之前的UPNP不正常的问题是不是超频引起的，如果是，我真是哭晕在厕所。2333333333333.

ywyha, 2020-11-28:
大佬，我还有一个情况，就是AC2100重启WAN口有时候会导致路由器死机，而且概率很高，三两回就需要拔电，不拔电的时候也会大概率导致路由器重启，我现在不知道本地连接或者异常重启的情况是否是超频引起，等我刷刷其他固件再来反馈下。其实如果可以出两版固件就好了，一个默认频率版本，一个OC超频版……

我之前用K2P LEAN的固件，半年多从来没有出现过本地连接断开的情况，而我最近一周已经出现过两种路由器三次异常本地连接断开的情况了，断开恢复后进路由器发现路由器异常重启了……

fatinghenji, 2021-02-01:
小米ac2100可以用？
是红米ac2100还是小米ac2100

ywyha, 2021-02-01:
> 小米ac2100可以用？
> 是红米ac2100还是小米ac2100

我说错了，是红米AC2100，我的红米体质很差，连1000hz都不稳定

closed 2021-12-29 by kiddin9