kiddin9/Kwrt #209: Phicomm K2P 升级到12.05.2020版本后 Wi-Fi无法启用

ideatechfr, 2020-12-05:
你好！
今天系统刷到12.05.2020版本后，Wi-Fi功能不能启动显示已禁用，另外我尝试手动设置了新的信道，无线安全选项里只有无加密一个选项。麻烦看看是什么问题. 

谢谢

ideatechfr, 2020-12-06:
更新到12.06.2020版本，Wi-Fi能激活使用

- 网络标签里有2个无线，第一个无线点击进入后显示设备未激活
[image]
- 访客网络功能无效，勾选启用及创建/删除 或单独勾选启用都发现不了访客网络

garypang13, 2020-12-06:
把/etc/config/wireless这个文件删掉,第一个无线菜单就会消失了.

ideatechfr, 2020-12-06:
> 把/etc/config/wireless这个文件删掉,第一个无线菜单就会消失了.

感谢，意思是把这个功能先屏蔽掉了，想知道这个功能在K2P上是否能正常运行？

ideatechfr, 2020-12-08:
版本更新到 12.09.2020 Wi-Fi 5G不能激活，麻烦大佬看一下
[image]

garypang13, 2020-12-09:
@ideatechfr 5G设置个信道就好了,

ideatechfr, 2020-12-09:
@garypang13 感谢大佬，麻烦问一下现在是刷12.06这个版本吗？（记得昨天看看到有个12.09这个版本）5g的问题把版本刷回到11.30了。

garypang13, 2020-12-09:
@ideatechfr 我在重新编译,默认设置好信道,

ideatechfr, 2020-12-09:
@garypang13 感谢!

closed 2021-01-07 by garypang13