kiddin9/Kwrt #231: 2020-12-31 04:17:13 Main Node: Can't find SSR program, start failed!

2607627233, 2020-12-31:
bypass 无法使用,只成功了一次之后就不能启动

2020-12-31 04:15:26 SSR-rules : Check network status...
2020-12-31 04:15:26 SSR-rules : Check network status successful!
2020-12-31 04:15:26 SSR-rules : Check IP/GFW files...
2020-12-31 04:15:26 SSR-rules : Download IP/GFW files...
2020-12-31 04:15:27 SSR-rules : Check IP/GFW files successful!
2020-12-31 04:15:29 SSR-rules : Start iptables rules successful!
2020-12-31 04:15:30 Main Node: ShadowsocksR 4 Threads Started!
2020-12-31 04:15:30 UDP Relay: ShadowsocksR Started!
2020-12-31 04:15:31 SmartDNS : Use https://cloudflare-dns.com/dns-query to start in DoH mode (Foreign DNS)
2020-12-31 04:15:31 SmartDNS : Use https://dns.alidns.com/dns-query to start in DoH mode (Domestic DNS)
2020-12-31 04:15:31 ChinaDNS : Start ChinaDNS-NG successful!
------------------------------------------------------------------------------------------------------------------------------------------------------------
2020-12-31 04:15:46 SSR-rules : Check network status...
2020-12-31 04:15:47 SSR-rules : Check network status successful!
2020-12-31 04:15:47 SSR-rules : Check IP/GFW files...
2020-12-31 04:15:47 SSR-rules : Check IP/GFW files successful!
2020-12-31 04:15:49 SSR-rules : Start iptables rules successful!
2020-12-31 04:15:49 Main Node: ShadowsocksR 4 Threads Started!
2020-12-31 04:15:49 UDP Relay: ShadowsocksR Started!
2020-12-31 04:15:49 SmartDNS : Use https://cloudflare-dns.com/dns-query to start in DoH mode (Foreign DNS)
2020-12-31 04:15:49 SmartDNS : Use https://dns.alidns.com/dns-query to start in DoH mode (Domestic DNS)
2020-12-31 04:15:49 ChinaDNS : Start ChinaDNS-NG successful!
------------------------------------------------------------------------------------------------------------------------------------------------------------
2020-12-31 04:17:09 SSR-rules : Check network status...
2020-12-31 04:17:09 SSR-rules : Check network status successful!
2020-12-31 04:17:09 SSR-rules : Check IP/GFW files...
2020-12-31 04:17:09 SSR-rules : Download IP/GFW files...
2020-12-31 04:17:10 SSR-rules : Check IP/GFW files successful!
2020-12-31 04:17:13 SSR-rules : Start iptables rules successful!
2020-12-31 04:17:13 Main Node: Can't find SSR program, start failed!
------------------------------------------------------------------------------------------------------------------------------------------------------------
2020-12-31 04:19:00 SSR-rules : Check network status...
2020-12-31 04:19:00 SSR-rules : Check network status successful!
2020-12-31 04:19:00 SSR-rules : Check IP/GFW files...
2020-12-31 04:19:00 SSR-rules : Check IP/GFW files successful!
2020-12-31 04:19:03 SSR-rules : Start iptables rules successful!
2020-12-31 04:19:03 Main Node: Can't find SSR program, start failed!

garypang13, 2020-12-31:
@2607627233 已修复,后台软件包中更新bypass即可.

2607627233, 2020-12-31:
> @2607627233 已修复,后台软件包中更新bypass即可.

重装之后能用，但是更新之后又不能用了。。。

closed 2021-01-07 by garypang13