kiddin9/Kwrt #246: X86:ppp-mod-pppol2tp无法安装

fhleo, 2021-01-08:
提示：
已安装的软件包 kernel 版本不兼容，要求 5.4.86-1-1cc178c1… 而 5.4.85-1-470d3e14… 已安装。
报错信息：
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-l2tp found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-pppol2tp found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pppol2tp:
 * 	kernel (= 5.4.86-1-1cc178c1f8d6c85a5513a864a5d45c6e)
 * opkg_install_cmd: Cannot install package ppp-mod-pppol2tp.

garypang13, 2021-01-08:
ppp l2tp 都属于淘汰的技术了,别用了,

fhleo, 2021-01-08:
请问目前 L2TP 客户端应该使用什么哪个包

garypang13, 2021-01-08:
@fhleo xl2tpd l2tpd 下一版支持.

closed 2021-12-29 by kiddin9