kiddin9/Kwrt #247: mariadb 全新安装配置完后启动不了

ethan-www, 2021-01-09:
全新安装，操作命令如下
opkg install mariadb-server mariadb-client
mkdir /mnt/vdb/mysql
chmod 777 /mnt/vdb/mysql
#sed -i 's#.*datadir.*#datadir='\/mnt\/vdb\/mysql#g' /etc/mysql/conf.d/50-server.cnf
uci set mysqld.general.enabled='1'
uci commit mysqld
mysql_install_db  --force
/etc/init.d/mysqld start

完成后报下面的错误
2021-01-10  0:45:44 0 [Warning] Could not open mysql.plugin table. Some options may be missing from the help text
2021-01-10  0:45:44 0 [Warning] Could not open mysql.plugin table. Some options may be missing from the help text
2021-01-10  0:45:44 0 [Warning] Could not open mysql.plugin table. Some options may be missing from the help text

之前同样命令步骤操作完后就能运行了，这次重新做系统后同样操作步骤，配置完成后启动不了

garypang13, 2021-01-09:
@ethan-www mariadb-common mariadb-server-base mariadb-server mariadb-client-base mariadb-client 都装上试试

ethan-www, 2021-01-09:
opkg install mariadb-server mariadb-client 就会把这几个依赖包自动安装上。
检查过了：mariadb-client mariadb-client-base mariadb-common mariadb-server mariadb-server-base
这些包都己经是装上了的

closed 2021-12-29 by kiddin9