kiddin9/Kwrt #258: 软件包里面的transmission和qbittorrent安装失败，安装ZeroTier后概况里面所有信息都获取不到了

3858809, 2021-01-14:
使用的设备是树莓派4B，昨晚折腾了老半天，软件包里面的transmission和qbittorrent安装失败 ，就想弄个下载工具一直装不上，大佬有没有空帮看一下。
安装了ZeroTier后概况里面的所有信息都获取不到了变成了？

garypang13, 2021-01-14:
安装失败报什么错

3858809, 2021-01-14:
> 安装失败报什么错

[image]

garypang13, 2021-01-14:
你这是安装transmission的报错吗,可是transmission并不依赖 python3的呀,

3858809, 2021-01-14:
我安装了transmission了也安装了中文插件，界面出现了但是是英文的，而且不能用，我想是不是缺了哪些依赖 然后我查了一下资料transmission好像需要安装
[image]
我在软件包里面搜索了 上图的插件点击安装了 就报上面的错误了

3858809, 2021-01-14:
> 你这是安装transmission的报错吗,可是transmission并不依赖 python3的呀,

后面我想transmission如果不能用那我就换 qBittorrent试一试  但是我发现 安装 qBittorrent 也是失败的 也是报上图的错 transmission能搜索出来很多插件和依赖 里面部分可以安装成功 部分安装失败 失败的，qBittorrent 就搜索出来两个可以安装的插件两个都安装失败 失败的报错都是上面截图那种错误

garypang13, 2021-01-14:
不好意思
那看来树莓派4B的软件包和R2S的不通用
树莓派的是 aarch64_cortex-a53
R2S的是 aarch64_generic/
都是aarch64我以为是通用的,看来只能下架树莓派4B了,不想再多维护个树莓派4B专用的软件库.

3858809, 2021-01-14:
> 不好意思
> 那看来树莓派4B的和和R2S的不通用
> 树莓派的是aarch64_cortex-a53
> R2S的是aarch64_generic /
> 都是aarch64我以为是通用的，看来只能下架树莓派4B了，不想再多维护个树莓派4B专用的软件库。
这样啊 真是太遗憾了T_T  很喜欢你的固件

closed 2021-12-29 by kiddin9